prusa3d/Prusa-Firmware #881: 3.3.0 RC Missing Show End stops

veggysaurus, 2018-06-28:
Flashed on to my MK3 today as having endstop issues and noticed that under Calibration menu the Show End stops is missing? has it moved?

Panayiotis-git, 2018-06-28:
Have you found any use of this functionality at a MK3? At a MK2 this function is useful because it has mechanical triggers. At MK3 only the z end stop is really helpful.

veggysaurus, 2018-06-28:
True, only the Z-stop is useful on mk3. It helps to diagnose if the PINDA is triggering. Even when sending marlins commands, x & y can’t be seen triggering via the printer interface.

PavelSindler, 2018-07-10:
Yes we removed this because there are no mechanical endstops on MK3. PINDa has LED diod so you can check its state without the menu. But it seems that many users used "show endstops" menu for checking PINDa state, so we will probably show Z endstop state in menu again. Thank you for feedback.

closed 2023-08-24 by 3d-gussner