prusa3d/Prusa-Firmware #882: Spurious characters appended to filename during print

LessPaul, 2018-06-28:
During a print additional characters will appear appended to the end of the filename.  Forum wisdom that it is caused by random noise, however I have seen these characters mirror the beginnings of other display strings which is highly unlikely to be caused by random noise. Possilby a stringlength error?

Mk3 Firmware 3.2.1  Filename: OneCorner2.gcode

Various characters were appended to filename during print, including but not limited to: #0, "8,  and 0 however it was the following that leads me to believe it is a string length error as the characters mirror the beginning of line 3:

[image]

Area5142, 2018-07-12:
This bug is also present in old firmware: 3.1.1-RC4 b143.

zerg960, 2018-07-26:
Still present in 3.3.1-845; saw the characters `!0` for a brief moment during a print. I've actually seen it a fair number of times with the 3.2.1 firmware as well.

I'm thinking its a hardware issue:
`#0` is a 3-bit error
`"8` is a 2-bit error
` 0` is a 1-bit error
`!0` is a 2-bit error
`$1  ` is a 2-bit error
moving `SD 9` to position 100 from 20 is a 3-bit error

I'd suggest changing:
```
void lcd_pulseEnable(void)
{
	digitalWrite(lcd_enable_pin, LOW);
	delayMicroseconds(1);    
	digitalWrite(lcd_enable_pin, HIGH);
	delayMicroseconds(1);    // enable pulse must be >450ns
	digitalWrite(lcd_enable_pin, LOW);
	delayMicroseconds(100);   // commands need > 37us to settle
}
``` in `blob/MK3/Firmware/lcd.cpp` to:

```
void lcd_pulseEnable(void)
{
	digitalWrite(lcd_enable_pin, LOW);
	delayMicroseconds(2);    
	digitalWrite(lcd_enable_pin, HIGH);
	delayMicroseconds(2);    // enable pulse must be >450ns
	digitalWrite(lcd_enable_pin, LOW);
	delayMicroseconds(98);   // commands need > 37us to settle
}```

to allow more time for the data line setup & hold times.

maximlevitsky, 2018-07-28:
I have the same issue. It was like that all the time since I assembled the printer. Today I tried to reset the cable, tried to route it a bit diffrently, including routing it outside all the other cables, but I still have the issue, which is cosmetic but annoying still.

I tried this change, but no luck.

leptun, 2019-10-27:
This issue should be fixed in one of my recent PRs. It is planned to be merged into 3.9

leptun, 2020-01-26:
This issue is currently fixed in the 3.9.0-Beta. There will be some changes to the code in that area in the future, but for now this issue is fixed. Would you mind closing this?

closed 2020-01-27 by LessPaul