prusa3d/Prusa-Firmware #884: Undesired, spurious "print fan error" e-stop at normal end of print

RetireeJay46, 2018-06-30:
Using the latest Prusa Slic3r, there is code generated at the end of the print with two M107 commands.
`M107 ; disable fan
; Filament-specific end gcode
G4 ; wait
M221 S100
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
`
This sequence of two M107's in a row causes the firmware to stop dead and report a PRINT FAN ERROR.  The nozzle is still at the location of the last point printed.  The nozzle heater is turned off, but the bed heat is still being controlled at the print's setpoint.  While it is fairly easy to cure the problem in Slic3r by removing the second M107 from the "Ending G-code", nevertheless it seems to me that the firmware should not generate an error in this circumstance.  And if it does truly encounter a PRINT FAN ERROR, the firmware should do a better job of handling the error, by raising the printhead away from the print and by turning off all the heaters.

PavelSindler, 2018-07-10:
When print fan error occurs, printer rises printhead away and turn off nozzle heater. It is the same as pause print and you can resume print from print fan error in menu (the same way as paused print). If you are printing from USB, printer turns off nozzle heater and sends pause action command (this is Octoprint feature). So if you are using octoprint you can define your own pause print sequence (rising print head etc.) and also resume print. We will check the behaviour which you described, print fan error should not occur in this scenario. Thank you for feedback.

RetireeJay46, 2018-07-10:
I print from USB, but I'm printing directly from a Windows computer running Repetier.  I don't have a Raspberry Pi and don't use Octoprint.  I like the Repetier interface and have been using it for years (on a different printer).  But I'm not aware of any provision in Repetier to respond to a "pause action command" coming from the printer.

jindrichbenes, 2018-07-10:
Could you please share a problematic gcode?

RetireeJay46, 2018-07-10:
I quoted the relevant parts of a g-code file in the original post.  But to answer your request, here is a complete file for the little cover that goes on the extruder body adjacent to the optical sensor.

[extruder-idler-plug.zip](https://github.com/prusa3d/Prusa-Firmware/files/2180621/extruder-idler-plug.zip)

RetireeJay46, 2018-07-16:
Well, this issue seems to be non-reproducible.

closed 2018-07-16 by RetireeJay46