prusa3d/Prusa-Firmware #885: No instructions on how to build the firmware after the new language support was added

maximlevitsky, 2018-06-30:
There are some scripts in 'lang' directory, and I sort of figured out how to run them, but I still get some missing strings in the built firmware (show up as '?').

Also the 'config.sh' script in this folder has hardcoded and windows only patches which should be fixed.

3d-gussner, 2018-08-02:
@XPila when we can expect some how-to? only i can see  is https://github.com/prusa3d/Prusa-Firmware/blob/MK3/lang/readme.txt

Here a Google translate try:
New support for more language in firmware

Changes to the existing frame:
1. Declaring localized text directly in the code, it is not necessary to keep the tables.
2. The two-language version (en_cz, en_de, etc.). Possibility to extend to multilingual (en_cz_de - for MK2).
3. Possibility of selecting the second language stored in the SPI flash (it will not be used in internal flash, only MK3).
5. Bash Postbuild Process for Perl script + tools for word processing.

Description:
The new framework is a bit similar to the i18n20, but suited to the AVR atmega for maximum simplicity and internal flashes.
Strings to be transferred are declared by special makers, the remainder obstara postbuild.
All localized texts are in a special section, in AVR case the strings must be located in the lower 64kB flash - called 'progm'.
After arduin build, only English will work, postbuilds need to be added to the dictionary to add the secondary tongue data and create the modified hexfile.
The only data that needs to be kept is translation dictionaries. These are text files where English text with translated text is always crawled.
Every text in a dictionary is one line, it can contain special characters in hexadecimal form (eg '\ x0a'). The following line is always translated text.
This simple format is chosen so that dictionary and translation can be managed only by git and several scripts.

If we add or change some text in the code, change the file lang_en_code.txt after you build and run 'update.sh'.
This is a generated file that contains all the localized texts used in the alphabetical code.
In git, we see the change that we manually transfer to the lang_en_xx.txt dictionary, creating a request for translation or proofreading of the requested text.
If we add or change some text in the word, change the lang_en_dict.txt file after running 'update.sh'.
It contains all localized texts from the dictionary (in English), respectively a set of their unification.
Ideally, the lang_en_code.txt and lang_en_dict.txt files should be the same.
If the dictionary is changed, binary files must be generated lang_en_xx.bin.

Use in the code, examples:

1. localized text declaration in code - macro '_i':
puts_P (_i ("Kill all people!"); // v cz write "Kill all the people!"

2. Declaring localized text as global constants - macro 'PROGMEM_I1' and 'ISTR':
const char MSG_PREHEAT [] PROGMEM_I1 = ISTR ("Preheat"); //declaration
puts_P (get_translation (MSG_PREHEAT)); // v cz list "Predehrev"

3rd operation get_translation - abbreviation macro '_T':
puts_P (_T (MSG_PREHEAT)); // v cz list "Predehrev"

4. declaration of localized text as local variable - macro '_I':
const char * text = preheat? _I ("Preheat"): _ I ("Cooldown");
puts_P (_T (text)); // v cz list "Predehrev" or "Chill"

5th declaration of unlocated text - macro 'PROGMEM_N1' and '_n' or '_N':
const char MSG_MK3 [] PROGMEM_N1 = "MK3"; //declaration
const char * text = _n ("MK3");
 or
const char * text = _N ("MK3");

3d-gussner, 2018-08-12:
@XPila Any update?

maximlevitsky, 2018-12-22:
Closing that as finally instructions are released

closed 2018-12-22 by maximlevitsky