prusa3d/Prusa-Firmware #887: 3.3.0 RC1 Nozzle crashed at „blob-cleaning-point“

RacingHell, 2018-07-02:
After several tries to start a print correctly z-axis moved all the way down without crash detection on full torque at the beginning of bed leveling. Bed was bend like 10mm down, needed to reset the printer to move this collision free. What in detail has been done befor this misbehaviour I can‘t tell anymore. PINDA detected.

RacingHell, 2018-07-02:
[image]

In the middle the negative surface of the nozzle at collision position.

jayPhantomic, 2018-07-05:
I had this happen in 3.2.1 as well. Just upgraded to 3.3.0, will let you all know if it does it again.

andredigenova, 2018-07-18:
Just had this happen again in 3.3.1. I moved Z really high up using the LCD interface and then initiated bed levelling. When going to probe the first point it drove the nozzle straight in to the bed hard.

O-Alex-O, 2018-09-23:
Happened to me today, with firmware 3.4.0
I started printing from sd-card, then I pressed the lcd-button to move the z-axis upwards
After preheating, he collides with the bed at the same point instead of performing the calibration.

closed 2023-09-04 by github-actions[bot]