prusa3d/Prusa-Firmware #890: MK3 3.3.0 Display shows blank screen for 7 seconds on reset

stahlfabrik, 2018-07-02:
After I updated to 3.3.0, on a printer reset, the screen goes blank after showing the spash screen "Original Prusa...". Then it shows nothing for 7 seconds. Then it shows again "Original Prusa..." and finally it is done.

The seven second pause looks like a timeout on error or something. It was never like that before updating to the 3.3.0 firmware

stahlfabrik, 2018-07-02:
Is it normal to wait for 7 seconds now? Or is that just because #891 - the flash cannot be read maybe?

stahlfabrik, 2018-07-02:
#891 is fixed after a third install try.

So my EINSY seams not broken and I will see and hear from others, if the 7 seconds are common - or where just happening because the failed installations.

closed 2018-07-02 by stahlfabrik

Panayiotis-git, 2018-07-02:
I have this delay also, but only after reset. Not after power-down and power-up.

Zaf9670, 2018-07-02:
I believe I have the issue on startup and reset. I will check when I get
home today.

Zaf9670, 2018-07-03:
My firmware upgrade completed without any issues as well. But I can confirm that mine does the blank screen after reset and power on. It shows 2 rows of block characters and then eventually shows the Prusa boot screen.

It is a little inconvenient but not a major issue at the moment.

maximlevitsky, 2018-07-03:
Since I dug a bit into the new language support, this due to how it is implemented. Since they don't want us to flash the bootloader (this can soft brick the board, will need ICP stuff to unbrick), they put another 'bootloader' in the main firmware which waits for few seconds for a magic command from a modified avrdude and if it gets it, it flashes the external flash chip. A bit sad that flash space is wasted for the secondary bootloader.

bubnikv, 2018-07-03:
> A bit sad that flash space is wasted for the secondary bootloader.

The secondary bootloader is really minimal.

maximlevitsky, 2018-07-03:
@Panayiotis-git yep.
@bubnikv Agree and I don't think this can be done better.
We only need some text message during this wait

maximlevitsky, 2018-07-03:
No, no, this only happens on re-boot, and  not on cold boot. Just like stock arduino.

stahlfabrik, 2018-07-03:
To answer my question - I did a cold start and I did a reset: NO - it does take the same time as before (I have selected English - if that makes a difference)

By the way: I closed the issue yesterday already:-) Because I think I witnessed that issue only because my installation failed. After the third installation everything works as expected!

hoppke, 2018-07-03:
Hmm. So for me 3.2.1 powers on/resets faster than 3.3.0 (I downgraded to 3.2.1 and back to 3.3.0 to confirm). The difference isn't massive (my internal clock tells me it's maybe +3s on top of what it used to take?), and it seems during this extra time it now displays 2 lines of blocks.

I initially flashed via octoprint, but then flashed back to 3.2.1 and re-flashed 3.3.0 with the enhanced slic3r - didn't seem to make a difference. Printer set to English.

@stahlfabrik, I really hope 3.3.0 made your mk3 also start up/reset a few seconds slower, and you just haven't picked it up :) And it's not schadenfreude on my part, it's just that the alternative would be more worrying (some 3.3.0 upgrade path bug?)

BTW, in terms of flashing feedback -- both octoprint and the windows slic3r beta were fine for me in that they both always displayed some progress message/bar in the app and on the printer's LCD, until flashing was done.

Panayiotis-git, 2018-07-03:
Even though the previous upgrade was OK, I re-flashed the 3.3.0-830 again. Also, I did a full reset and selected the English language.
Measurements:
Boot time from power on: 4secs
Boot time after reset: 10secs

@stahlfabrik I cannot understand why you measure the same time.

What @maximlevitsky said earlier, seems to be the reason. Not a big deal, but I wish the delay was smaller.

hoppke, 2018-07-03:
I think I'd prefer maybe multiple firmware images (with different lang sets). Insisting on a single uber-image that supports all languages at the same time seems... unusual, given how unlikely it is that an average user will care about having all language variants available at all times in the firmware.

matthew-humphrey, 2018-07-04:
It's probably more about Prusa engineering and support impact than it is user convenience. Personally what I would like is a controller board using a 32-bit micro, with much more flash this hack wouldn't be needed.