prusa3d/Prusa-Firmware #891: MK3 3.3.0 Only English and Cestina language are available

stahlfabrik, 2018-07-02:
I updated via Slicer 1.40.1 beta for Mac. I updated even twice.

The first time, I did a factory reset after the update and chose to erase all data. I was greeted with a language selection of "English" and "?".
So I updated the firmware again and made the factory reset again. This time, it is "English" and "Cestina" - I was expecting to see a bunch of firmwares?

Is that related to my other issue #890? I got the EINSY 1.1a.

stahlfabrik, 2018-07-02:
I updated a third time. Now it shows all languages in the settings menu under languages. Also after a factory reset. My slic3r hang after the second update try. I rebooted my Mac for the third flash. That worked now... So, @bubnikv something got stuck when I updated the firmware - but to be honest - maybe the blank screen at the end of the firmware update on the printer made me hit "x" - so I canceled flashing of the external flash? Maybe it would be good to print "Please wait" instead of a blank screen!

stahlfabrik, 2018-07-02:
So I close this issue - but the "blank screen" really should say "PLEASE WAIT" in the last seconds of install

closed 2018-07-02 by stahlfabrik

bubnikv, 2018-07-02:
> Maybe it would be good to print "Please wait" instead of a blank screen!

Blank screen on the printer? Or do you think the Slic3r firmware updater
should be more descriptive?

I wonder why did you cancel Slic3r firmware updater, if the progress bar
was in the middle. Is that correct?

stahlfabrik, 2018-07-03:
@bubnikv

I think what happened was that I focused my attention on the printers screen
There, the writing progress is shown from 0 to 100 percentage. 
Then a verify progress is shown from 0 to 100 percent. Then the screen on the printer goes blank. Here I am quite sure that I got nervous and hit the X button. 

On my third and successful attempt to install I saw that the blank screen on the printer falls into the period where the external flash is written to. 
So it would helped me to display a please wait during the last phase of the update. Also i might have benefited from a warning before the start of the update process that there is a new waiting period in the end where the screen stays blank. 

Best regards!

stahlfabrik, 2018-07-03:
Also the slicer firmware update process got into an endless loop running in the background, maxing out my cpu when I interrupted the update the first time. On Mac OS. The fans spun up. The slicer icon sat in the dock, unstoppable. I rebooted the Mac. 
Maybe you could harden the update process against interruptions, so that the process gets killed when the window gets closed.

bubnikv, 2018-07-03:
> On my third and successful attempt to install I saw that the blank screen on the printer falls into the period where the external flash is written to.

We should update the screen during the language flashing.

> Also i might have benefited from a warning before the start of the update process that there is a new waiting period in the end where the screen stays blank.

Please when discussing this issue, make it explicit, whether you mean the printer screen or the Slic3r :-) I suppose here you mean to update the printer screen with the warning.

I agree it would be great to make the update process more explicit about what it is doing. There are periods of reboot, which will blank the screen, but we may add some notifications of what will be coming.

> Also the slicer firmware update process got into an endless loop running in the background, maxing out my cpu when I interrupted the update the first time. On Mac OS. The fans spun up. The slicer icon sat in the dock, unstoppable. I rebooted the Mac.

@FidelCapo  @VojtechKral Please try to reproduce the firmware updater hanging on OSX when interrupted. Thanks.

stahlfabrik, 2018-07-03:
Sorry that I did not make myself clear. I was typing quickly on my Phone.

The blank screen on the printer really should not be blank - if possible - because I hit X button after a few seconds - I guess other users might so too.

So I think - on the printer and especially in the update-tool - the user should be advised on what to expect - so if the blank screen has to be there, I would advise you to prepare the user for that blank screen. IMHO - I am not the dumbest user around (LOL) - so I find it remarkable that I ran into that trap.

I want to add about by Mac OS specific bug report: The fans of the Mac spun up in order to cool down the CPU that was processing the hanging update-process - obviously I do not mean the printer's fans:-) Just to be clear:-)