prusa3d/Prusa-Firmware #893: unable to compile mk3 version

Patrik-svobodik, 2018-07-04:
Hi, I can't compile the mk3 version of the fw, the mk2 version is fine, but the mk3 version throws many errors, for example:

Arduino: 1.8.5 (Mac OS X), Vývojová deska: "RAMBo"

Archiving built core (caching) in: /var/folders/3g/7hcj_8917tj_j_b5ctn_mfjr0000gn/T/arduino_cache_503613/core/core_rambo_avr_rambo_0d41fba9067e14b890e82014de61d795.a
sketch/Marlin_main.cpp: In function 'gcode_M45.constprop':
Marlin_main.cpp:3072: error: unrecognizable insn:
 }
 ^
(insn 820 819 821 28 (set (reg:QI 962)
        (subreg:QI (mem/v/c:SI (plus:HI (reg/f:HI 961)
                    (const_int 8 [0x8])) [6 count_position+8 S4 A8]) 3)) sketch/xyzcal.cpp:773 -1
     (nil))
sketch/Marlin_main.cpp:3072:1: internal compiler error: in extract_insn, at recog.c:2202
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
lto-wrapper: /Users/patriksvoboda/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc returned 1 exit status
/Users/patriksvoboda/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
exit status 1
unrecognizable insn:

Panayiotis-git, 2018-07-04:
Check
https://github.com/prusa3d/Prusa-Firmware/issues/614#issuecomment-385034066

Patrik-svobodik, 2018-07-04:
Thanks, it fixed it

closed 2018-07-04 by Patrik-svobodik