prusa3d/Prusa-Firmware #895: MK2.5 Firmware 3.3.0 - Failing Calibration on 3rd Point

pshankland, 2018-07-04:
As the description; newly built MK2.5 from MK2S and updated immediately to 3.3.0 as instructed. Each time the wizard tried to find the 3rd calibration point it failed. Downgraded to 3.2.1, ran the wizard and all passed.

aderusha, 2018-07-05:
I'm seeing similar behavior.  I have a MK2 > MK2s > MK2.5, been running without problem on 3.2.1, upgraded firmware to 3.3.0 and calibration fails on the second calibration point.  Revert to 3.2.1 and everything works like normal again and prints complete without problems.

blazemi9, 2018-07-14:
I had the same problem. It helped me when I came up right side my heatbed with the pads. New XYZ calibration seems to have a limited height range.

clearbucketLabs, 2018-07-23:
Same here, after 5-6 tries the same thing.  since i was adjusting a few things on my X i figured i messed something up, but nope.   only downgrading helped.

morphogenetic2, 2018-08-20:
Same here, using 3.3.1 in MK3.  Nozzle pushes on the heatbed on the first point, and then crashes on the third point.  I had to downgrade to 3.1.2 to calibrate and then get back to 3.3.1