prusa3d/Prusa-Firmware #896: MK3 Flashing 3.3.0 firmware results in: "External SPI flash W25X20CL not responding" error

Petclaud, 2018-07-04:
Hi, just wanted to report that on my MK3, flashing the firmware to the latest 3.3.0 release results in the error as per the title. I have made several attempts with the same result. Flashing it to the 3.2.1 gives no errors. Trying again the 3.3.0 = same result. 

[image]

I see also already an entry on the forum describing the same issue: https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/firmware-3-3-0-for-original-prusa-i3-mk3-t22743-s20.html

adlamy, 2018-07-06:
Hi, 
If it can help, the link to my post about this issue on the forum
https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/firmware-3-3-0-for-original-prusa-i3-mk3-t22743-s40.html#p96454

Regards.

Petclaud, 2018-07-15:
Update: Also flashing with Slic3r 1.40.1 results in the error (which makes sense since the 1.40.1 RC2 is supposedly to be the same as the final release). 

I just had a chat session with Prusa's Tech Suppport and they promised to look into this.

lewnewby, 2018-07-19:
This occurs on the 3.3.1 release of the firmware as well.

ChrisKrieg, 2018-07-22:
Just had this happen to me when uploading 3.3.1 to the Mk3. Cannot perform a factory reset. Will try to upload firmware 3.2.1

tomlask, 2018-07-23:
I have the same problem. I got 2 MK3 printers. On one upgrade loads without issues, on second got the same error. Boards in both printers looks the same to me. I had to load and stay on 3.2.1 on one machine.

tomlask, 2018-07-23:
I took closer look on board with firmware upgrade issues.
The problem is... there's no W25X20CL chip on board!
These boards were produced in several versions, 
I got one referenced by assembly manual as version A (without add-on board).
Take closer look at the left side and compare it with that pic: https://ultimachine.com/products/einsy-rambo-1-1
These 3 chips near Z-axis connectors are U12 - U14 on schematic: https://github.com/ultimachine/Einsy-Rambo/blob/1.1a/board/Project%20Outputs/Schematic%20Prints_Einsy%20Rambo_1.1a.PDF
Sheet 5 on bottom of page.
So this is hardware issue - probably first versions were produced without 2Mb SPI flash.
If you have R-packs (resistors) instead of chips and no add-on board the new firmware would not load.

Board with add-on has 3 chips on it, and I'm quite sure these are missing buffers and SPI Flash.
There were no problem since deevlopers started using SPI Flash ;-)

So I need to contact suport for add-on or replacement.

Petclaud, 2018-07-23:
Hmm if that is true, it explains a lot. Thx for sharing this info 👍 . I guess now Prusa will need to see who has received versions A boards and who another version and furnish replacement boards for the "A-users' or avoid SPI flash.

lewnewby, 2018-07-25:
I found out what caused the issue for me, I'm an idiot and removed the "Daughter board" from my Einsy quite come time ago to make room in the case and the missing chips are actually on it. This wasn't an issue until FW 3.3.0 but now it is so i have reinstalled the daughter board and all is working well.

adlamy, 2018-07-25:
This confirm the issue only with A version Einsy board that comes with no daughter board.

dikodahan, 2018-07-25:
I also have the board version without the daughter card and am experiencing the same issue.

bubnikv, 2018-07-25:
Please contact the support, they shall be informed about this particular issue.

Petclaud, 2018-07-25:
Just had a chat with Prusa. They ask for a photo of the board and if proven to be a type A board, than they will ship a 'patch board'. In the electronics assembly manual it is here: https://manual.prusa…ics+assembly/513?lang=en#s8303

closed 2018-07-25 by Petclaud