prusa3d/Prusa-Firmware #901: MK3 Firmware 3.3.0 -  Filament is not counting while printing

gdombiak, 2018-07-07:
Many people are reporting that the filament counter is not "counting" after the update to 3.3.0. See: https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/firmware-3-3-0-for-original-prusa-i3-mk3-t22743-s50.html#p96594

Olefff, 2018-07-09:
Confirm I have this issue too on 3.3.0

cnlson, 2018-07-09:
I also have the issue
[image]

PavelSindler, 2018-07-10:
It don't work during print, but statistics are shown corretly after print is finished.

PavelSindler, 2018-07-16:
Fixed in PR https://github.com/prusa3d/Prusa-Firmware/pull/913 and released in [3.3.1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.3.1).

closed 2018-07-16 by PavelSindler