prusa3d/Prusa-Firmware #902: FW3.3.0-830 on Mk3 does not recover from power failure

bobstro, 2018-07-08:
I've had trouble with resuming prints after power outages since FW3.2.0. Prior to that, it worked quite reliably. I updated to FW3.3.0 today and tried again. Instead of the prompt to resume, I immediately got a "resuming print" message instead of the "resume print?" prompt in earlier versions. The printer waited for the bed & nozzle to heat up, then moved the extruder into position. It then now stalled displaying a "Bed done" message. The extruder & bed remained at the correct temperatures.

PavelSindler, 2018-07-10:
"resume print?" is shown in case that it was long power failure. If power failure was short, printer auomaticly resumes print. Was it SD card print? What was your filename? Was this file stored in subfolder?

bobstro, 2018-07-10:
It was a brief outage (plug pull). I can reproduce longer if desired. Definitely an SD card print, as I don't use Octoprint. Longer filename with spaces. Any particular pattern you'd like to test for filename?

Zaf9670, 2018-07-10:
Wasn't plug pulling the improper method for testing? you should be using an external switch like a power strip or light switch for testing.

Plug pulling and powering off the PSU are not what the power panic can easily detect IIRC.

bobstro, 2018-07-10:
Power off via switch defeats recovery (at least according to what I've read). I've also been using a power strip with switch. Have tried both ways, but will repeat. This is pretty obvious though.

bobstro, 2018-07-15:
Are you awaiting more of a response? I think I've included everything.

closed 2019-02-04 by bobstro