prusa3d/Prusa-Firmware #903: Auto pre-head on filament load/unload

tubbers, 2018-07-08:
When you select filament unload or load when the printer is cold you get an error message that the nozzle is cold. You then have to go through multiple items to preheat it and then come back and try again. Another i3 clone I used would simply ask if you want to auto-preheat the nozzle when you select load or unload from a cold state. It would make the user experience simpler and faster for not disadvantage that i can see. The preheat could be automatic on selecting load or unload, or it could give you the option with one click. The model I had used a default temp of 230, probably ok for place, abs and petg. More exotics would need to be manual.

Panayiotis-git, 2018-07-08:
Nice idea. Or, instead of just displaying the message, it should give the choice to go directly to the preheat menu. Also, it would be great to give preheat choices only for the hotend.
I usually find myself navigating to the setting>set temp menu and rotate like a maniac the button to reach the desired temperature.

DataCrypt, 2018-07-16:
I agree with this request.  The preheats are ONLY for Prusa filaments and I constantly have to go to the sub menus to select the temperature to preheat to and then load the filament.  It should at least jump us to the preheat menu automatically.

dougshell, 2021-03-20:
This is still an issue.

How is this still an issue?

closed 2021-05-20 by JohnnyDeer