prusa3d/Prusa-Firmware #904: Blobs upon resume after M600

mgg4, 2018-07-08:
Environment:  Prusa i3 MK3
Firmware: 3.3.0

Problem:
When issuing an M600 to cause a filament change, the print head moves to the front right.  The filament change completes normally.  However, there are some additional extrusion/retraction instructions after the filament load is completed, and before the print resumes.  These commands are found on lines 6573 to 6579 of Marlin_main.cpp.  The issue is the amounts extruded are set using constants, and there is no way for a user to control these operations.  

Desired outcome:
This final priming should use the values set in the M600 line if specified, and use the constants as the defaults.  In this way, the user can configure the amount of filament extruded and retracted in this final step.

Area5142, 2018-07-12:
This maybe related to #197 
I find it very annoying with the extra extrude after Color Change M600 - I have to be fast and lucky to catch the extra filament before it deposits on the part...
I would suggest that there is no extra extrude, except if a parameter is given to the M600 command, as @mgg4 suggest.

Area5142, 2018-07-16:
@HFMan I know this behavior is intentionally from #197, but it still gives me problems. When printing PETG the color is seldom clear after first purge and I normal purge 2-3 times to get a clean color.
If I don't remove the purged filament as it comes out or is at a position with lots of space under the nozzle, it will often reach the nozzle or heater block and stick to it.
Molten PETG is very sticky and just a little amount around the nozzle will ruin the rest of the print (stringing and dragging the surface).

Is it really necessary to flow more filament through the nozzle right after a purge.
Somewhere it was mentioned that the purpose was to build up the pressure in the nozzle, but that is now handled with LA (Linear Advance) in the firmware.

tbrummel, 2018-09-06:
I also believe the extra extrusion should be removed.  The color change extrudes sufficient filament to continue the print and the unknown z and extrusion amount required to 'leave the extra on a pile' is a real problem.  It rarely works as HRMan describes.  Maybe an option supported by GCODE would allow configuration by Slic3r.

St0RM53, 2019-09-20:
I made a post in the forums regarding this issue because i am doing multimaterial multicolor prints with a stock mk3s (no mmu, single extruder)
Post with details and video here: https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/m600-is-broken/
Is this ever gonna be "fixed"?

St0RM53, 2019-12-17:
Still not fixed!?!?

GunSmoker, 2020-04-30:
It is not a bug. If you are removing filament from nozzle - you are doing it wrong. You are supposed to grab extruded filament with tweezers and hold it. Hot end will prime and move to resume. Moving hotend will break filament, leaving clean primed nozzle, and filament blob in tweezers.

closed 2023-09-06 by github-actions[bot]