prusa3d/Prusa-Firmware #905: Enable Power Panic in a paused print

bestgustavo30, 2018-07-09:
Hi, I paused a print for a minute of two and then had a blackout for approx 3 minutes and after the blackout the printer turned on normally and forgot the past print so I lost the paused print

### Behavior
1. Start a print
2. Pause the print
3. Disconnect the printer, or suffer a blackout
4. Reconnect the printer
   - Results:
     - The printer turns on normally, forgetting the paused print

   - Expected Results
      - The printer will turn on, and display a message of the paused and unfinished print
      - Or, if the blackout was only a couple of seconds only wait to the user to resume the paused print, if the bed is already cool prompt the same message of a common long blackout.

Is this a new feature request?
Yes 
or maybe a bug I dont know

Thanks :)

PavelSindler, 2018-07-10:
Thank you for feedback. We are aware that this should be improved.

TheCutter, 2018-07-16:
Yes, we definitely need this feature. I wonder why power panic doesn't work with paused prints. Should be easier and better to handle then during a print. Please add this! :D

eliotik, 2020-02-05:
Hi, happened to me recently. Any update around this feature?

leptun, 2020-02-05:
There have been major changes to the PowerPanic implementation in the last week or so. Many changes are planned for the 3.9.0-RC1 which is around the corner. It does not handle a paused print as of yet, but in the near future this is planned to be fixed.

eliotik, 2020-02-05:
Thank you for update @leptun :)

closed 2023-07-18 by DRracer

gusdleon, 2023-07-18:
❤️💯