prusa3d/Prusa-Firmware #910: flashing 3.3.0 to MK3 verification error

tomaslieven, 2018-07-10:
i flashed 3.3.0 via slic3r 1.40.1+ to my MK3 but that resulted in first mismatch at byte 0x3c001
         0x00 != 0x06

`

avrdude: Version 6.3-20160220-prusa3d, compiled on Jul  5 2018 at 06:36:32
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Applications/Slic3r.app/Contents/MacOS/../Resources/avrdude/avrdude.conf"

         Using Port                    : /dev/cu.usbmodem1411
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "/Users/..../Downloads/prusa3d_fw_MK3_3_3_0.hex"
avrdude: writing flash (253106 bytes):
avrdude: stk500v2_command(): command failed

Writing | ################################################## | 100% 44.30s

avrdude: 253106 bytes of flash written
avrdude: verifying flash memory against /Users/.../Downloads/prusa3d_fw_MK3_3_3_0.hex:
avrdude: load data flash data from input file /Users/.../Downloads/prusa3d_fw_MK3_3_3_0.hex:
avrdude: input file /Users/.../Downloads/prusa3d_fw_MK3_3_3_0.hex contains 253106 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 32.41s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x3c001
         0x00 != 0x06
avrdude: verification error; content mismatch

avrdude done.  Thank you.
`

i also tried to downgrade to 3.2.1 with slic3r and FirmwareUpdaterV21b which i had on mac with same error. 

printer now loads with flickering screen and i cant use that

tomaslieven, 2018-07-11:
only way to get printer back working was to reflash and i tried all the way back to 3.1.3 with FirmwareUpdaterV21b which made printer work again

tomaslieven, 2018-07-11:
also flashing from 3.1.3 to 3.2.0 with FirmwareUpdaterV21b is causing error:
[image]

GurliGebis, 2018-07-11:
Sounds like the flash chip is going bad. You might want to contact support via the chat on the website. The flash shouldn't wear out that fast, especially for a microcontroller

XPila, 2018-07-16:
@tomaslieven it seems to be flash chip or bootloader problem, have you original prusa bootloader?

Karlosek, 2018-07-16:
I have the same problem.
The last fw I can flash via FirmwareUpdaterV21b is 3.1.1 stock fw from download website. I tried different USB cables, no luck.

This is the output when I was trying to flash the latest fw via slic3r:

Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "/Users/kk/Desktop/prusa3d_fw_MK3_3_3_0.hex"
avrdude: writing flash (253106 bytes):
avrdude: stk500v2_command(): command failed

Writing | ################################################## | 100% 44.33s

avrdude: 253106 bytes of flash written
avrdude: verifying flash memory against /Users/kk/Desktop/prusa3d_fw_MK3_3_3_0.hex:
avrdude: load data flash data from input file /Users/kk/Desktop/prusa3d_fw_MK3_3_3_0.hex:
avrdude: input file /Users/kk/Desktop/prusa3d_fw_MK3_3_3_0.hex contains 253106 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 32.31s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x3c000
         0x10 != 0x14
avrdude: verification error; content mismatch

avrdude done.  Thank you.

[image]

Karlosek, 2018-07-16:
My Einsy came directly from Ultimachine.

Karlosek, 2018-07-16:
I am normaly on MAC, but just tried to flash the fw using PC. No luck either. The same error.

3d-gussner, 2018-07-16:
@Karlosek stock Ultimachine Einsy has standard Arduino bootloader and Prusa firmware may not fit. That's why Prusa has its one boot loader saving few bytes.

Karlosek, 2018-07-16:
Yeah, might be the problem. How easy to reflash the bootloader?

3d-gussner, 2018-07-17:
@Karlosek search the github issues there are some helpful comments. @XPila Is it possible you provide a how-to flash prusa boot loader on Einsy ?

XPila, 2018-07-17:
bootloader update is possible with USBasp adapter or similar https://www.fischl.de/usbasp/

XPila, 2018-07-17:
next release should be <0x3c000 and compatible with this bootloader, but we cannot assure this in future...

Karlosek, 2018-07-17:
All I have right now is USBTiny or Arduino Uno somewhere.  I asume I can flash the bootloader with USBTiny, right?

tomaslieven, 2018-07-17:
@XPila yeah i have orginal prusa MK3 printer no changes or hardware upgrades

Karlosek, 2018-07-17:
I am trying to flash the bootloader with Arduino Uno. It says:

Arduino: 1.8.5 (Mac OS X), Board: "RAMBo"

Error while burning bootloader: missing 'bootloader.tool' configuration parameter

I understand that RAMBo parameters (processor, fuses etc) has to be mentioned in boards.txt but I cannot find the correct config. Any idea?

XPila, 2018-07-18:
@tomaslieven if you have stock MK3, please contact Prusa3D support, you will receive new Rambo board.

Karlosek, 2018-07-18:
Thank you XPila. I tried flashing the bootloader with Arduino UNO and AVRdude from command line. I found some info on Ultimachine GitHub.:

2560 upload command:
$ avrdude -v -v -p m2560 -c avrispmkII -P usb: -e -Ulock:w:0x0F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xD0:m -Ulfuse:w:0xFF:m -U ./stk500boot_v2_mega2560.hex 

-You might need to be super user.
-The last -U flag is the location of the hex file you want to upload.

https://github.com/ultimachine/RAMBo-Uploader/blob/master/notes.txt

But still no luck. I might as well go to Prusa "factory" and ask for help there. I hope they can flash the board for me. (I live in Prague). Might be faster than order dedicated ISP programmer.

Barnabasek, 2018-07-21:
What version of the Einsy board do you have? Is there a Prusa "patch board" inserted in the LCD connectors?

Karlosek, 2018-07-22:
[image]

Thank you for your comment!  I connected it this way - exactly how you describe it. thanks for the command. I will try it tonight.

I visited Prusa and I am pretty sure the service technician connected his USBasp dongle in the middle of J19 (pin 5,6,7,8,9,10) where the round sticker was. I just hope he did not burn anything. No need to mention he could not help me.

neiion, 2018-07-22:
Hi
I am in need of help, Here is what i have done so far. maybe i am doing it completely wrong?
I have connected the einsy board (from ultimaker) to the arduino( using connection diagram from above ). i then loaded audrino isp sketch to arduino. I then open cmd in c:\winavr bin folder, copied prusa bootloader hex file into bin folder. I then run " avrdude -v -v -c avrisp -b 19200 -P com23 -pm2560 -Uefuse:w:0xFD:m -Uhfuse:w:0xD0:m -Ulfuse:w:0xFF:m -Ulock:w:0x0F:m -Uflash:w:stk500boot_v2_mega2560.hex:i "  It keeps giving me a "Yikes! invalid device signature error" .

Am i going about this the right way?  any help would be great

Thanks

neiion, 2018-07-22:
Thanks for the info KoKuToru 

I am using the arduino uno as the programmer running arduino as isp sketch.  I am lost as to where i run the commands to upload bootloader to einsy connected to arduino. is it from winavr or from inside the arduino ide. thanks for the help

Karlosek, 2018-07-22:
I just tried the command and got error. Something with the fuses.

Karels-MBP:~ kk$ avrdude -v -v -c avrisp -b 19200 -P /dev/tty.usbmodem1411 -pm2560 -Uefuse:w:0xFD:m -Uhfuse:w:0xD0:m -Ulfuse:w:0xFF:m -Ulock:w:0x0F:m -Uflash:w:/Users/kk/stk500boot_v2_mega2560.hex:i

avrdude: Version 6.3, compiled on Sep 18 2017 at 20:31:51
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/usr/local/Cellar/avrdude/6.3/etc/avrdude.conf"
         User configuration file is "/Users/kk/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/tty.usbmodem1411
         Using Programmer              : avrisp
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel AVR ISP

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x02

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
         Hardware Version: 32766
         Firmware Version: 32766.-406646912
         Topcard         : STK502
         Vtarget         : 1.8 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.

Karels-MBP:~ kk$

Karlosek, 2018-07-22:
I then used the command line with the -F param, to override the check:
avrdude -v -v -c avrisp -b 19200 -P /dev/tty.usbmodem1411 -pm2560 -F -Uefuse:w:0xFD:m -Uhfuse:w:0xD0:m -Ulfuse:w:0xFF:m -Ulock:w:0x0F:m -Uflash:w:/Users/kk/stk500boot_v2_mega2560.hex:i

Got this msg:

avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x000000 (retrying)
avrdude: Device signature = 0x000000 (retrying)
avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
avrdude: Expected signature for ATmega2560 is 1E 98 01
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.

KoKuToru, 2018-07-23:
Please don't `-F` There is a reason for these checks.. 

1. Get an arduino UNO
2. Open Arduino IDE
3. Load Arduino ISP sketch
6. Upload Sketch (Board = Arduino UNI, Programmer = AVRISP mkII)
7. connect ISCP to Einsy   
I used the lcd/sd cabel like this: 
[image]
[image]
[image]
[image]

As you can see, the moment you connect to Einsy the LED above the USB-Port should light up

8. open a terminal and execute avrdude
```
[luca@LucaArchLinux stk500v2-prusa-master]$ ls
avr_cpunames.h       command.h  lcd.h        Makefile    stk500boot.c  stk500boot.hex
[luca@LucaArchLinux stk500v2-prusa-master]$ ../arduino-1.6.8-linux64/arduino-1.6.8/hardware/tools/avr/bin/avrdude -C /home/luca/3dPrinter/arduino-1.6.8-linux64/arduino-1.6.8/hardware/tools/avr/etc/avrdude.conf -v -v -c avrisp -b 19200 -P /dev/ttyACM1 -pm2560 -Uefuse:w:0xFD:m -Uhfuse:w:0xD0:m -Ulfuse:w:0xFF:m -Ulock:w:0x0F:m -Uflash:w:stk500boot_v2_mega2560.hex:i

avrdude: Version 6.0.1, compiled on Apr 14 2015 at 19:04:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/home/luca/3dPrinter/arduino-1.6.8-linux64/arduino-1.6.8/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/luca/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM1
         Using Programmer              : avrisp
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel AVR ISP
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us
[... 72 lines cut ...]
avrdude: reading input file "stk500boot_v2_mega2560.hex"
avrdude: writing flash (258444 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 258444 bytes of flash written
avrdude: verifying flash memory against stk500boot_v2_mega2560.hex:
avrdude: load data flash data from input file stk500boot_v2_mega2560.hex:
avrdude: input file stk500boot_v2_mega2560.hex contains 258444 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 258444 bytes of flash verified

avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as D0
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (H:FD, E:D0, L:FF)

avrdude done.  Thank you.

[luca@LucaArchLinux stk500v2-prusa-master]$ 
```

Karlosek, 2018-07-23:
Thank you very much for detailed walkthrough!  I am currently on vacation for three weeks.  I will try it again when I come back home.

I really appreciate all your help!

neiion, 2018-07-24:
thanks for detailed info  . going to give it a try later on. cheers

baloan, 2018-08-26:
Another description with some more background information is at https://learn.sparkfun.com/tutorials/installing-an-arduino-bootloader. It was not clear to me that the Arduino needs to have the _ArduinoISP_ sketch uploaded which is included in the Ardunio IDE as an example.

Karlosek, 2019-02-04:
Just for a record: Arduino UNO as ISP programmer needs 10µF
[image]
 electrolytic capacitor connected to RESET and GND.

martins6413, 2019-03-09:
Hallo,ist dieser tread noch aktuell ?

closed 2023-09-06 by github-actions[bot]