prusa3d/Prusa-Firmware #920: Add M860 to Wiki - Supported G codes

kazibole, 2018-07-11:
The "M860 Sx" g-code (wait for PINDA probe temperature) is not listed in the Wiki @ https://github.com/prusa3d/Prusa-Firmware/wiki/Supported-G-codes

CrashCash, 2018-08-09:
And while you're at it, neither is M73 - the new "time remaining" code.

Prusa-Support, 2022-06-06:
Hello! Sorry for the very late response to this. We added these, but currently we refer to another page from our Knowledge Base for GCode list: https://help.prusa3d.com/article/prusa-specific-g-codes_112173 

Alessandro Pantaleo
Prusa Research

leptun, 2022-06-06:
To add to this answer, I think a very up to date gcode documentation can also be found here: https://reprap.org/wiki/G-code
We try to keep this wiki updated as well. You can also use doxygen on the source code (all gcodes are documented there)