prusa3d/Prusa-Firmware #924: 3.3.0 Disconnecting from OctoPrint after G28 command.

chatrat12, 2018-07-12:
Using latest stable octoprint (1.3.8). Every time I run a G28, the printer stops communicating with OctoPrint. More info can be found [here](https://github.com/foosel/OctoPrint/issues/2721) where I opened an issue on the OctoPrint repo.

kazibole, 2018-07-12:
Does the same still happen when you issue "G28 W" (home without mesh bed levelling) and then "G80" (mesh bed levelling)?

I've noticed the same thing on mine, running OctoPrint 1.3.9-rc2 and MK3 firmware 3.3.0, when printing from the SD card - it will timeout after the homing and mesh bed levelling. Although my gcode file has "G28 W" and "G80" instead of just "G28" and it still times out. However from my limited testing from the OctoPrint terminal, "G28" will timeout, but "G28 W" and "G80" will not.

foosel, 2018-07-12:
It looks like G28 is lacking the acknowledging ok at its end in 3.3.0 which might be causing this.

kazibole, 2018-07-12:
FYI I can successfully start and finish an SD card print, with OctoPrint monitoring, with the following gcode:

G28 W
G80
G81

The acknowledging ok is indeed missing from the G28 (or G80?, but is present on G28 W), but the G81 will give an acknowledging ok which makes it so that OctoPrint does not time out. Again this is from an SD card print.

PavelSindler, 2018-07-13:
Thank you for reporting this. Yes, G28 is missing ok. We will fix it and make patch release next week.

PavelSindler, 2018-07-16:
Fixed in PR https://github.com/prusa3d/Prusa-Firmware/pull/927 and released with [3.3.1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.3.1).

closed 2018-07-16 by PavelSindler