prusa3d/Prusa-Firmware #928: feature request: total number of prints appears in lcd_menu_statistics

johncronan, 2018-07-15:
Can there be an EEPROM variable that stores the number of prints, as for filament and runtime? This would be very helpful to use along with the 'fail stats' info. (And a related one: I would be interested to know how many prints were canceled by the user.)

Thanks

JohnOCFII, 2018-07-15:
> Can there be an EEPROM variable that stores the number of prints, as for filament and runtime? This would be very helpful to use along with the 'fail stats' info. (And a related one: I would be interested to know how many prints were canceled by the user.)

These items get to be "expensive" in terms of using up scarce resources in firmware.  If you were to use OctoPrint, you can track all of these items and more.  https://octoprint.org

johncronan, 2018-07-15:
Ok, forget the second one. Just one int, please!

(Users here definitely appreciate the simplicity of having just one machine interface, with an SD card.)

closed 2023-08-24 by 3d-gussner