prusa3d/Prusa-Firmware #931: 3.3.0-830 Mk2.5 Does not show used filament under statistics while printing

stsfin, 2018-07-17:
3.3.0-830 Mk2.5 Does not show used filament under statistics while printing the counter stays on zero. Toggling the filament sensor on or off does not do anything.

Also printtime counts downwards from the ETA, and does not show the time that has passed.
[image]

PavelSindler, 2018-07-17:
There was bug in filament used. It was fixed in PR https://github.com/prusa3d/Prusa-Firmware/pull/919 and is present in 3.3.1 firmware version. We will look closer on print time. Thank you for reporting this.

mkbel, 2018-09-19:
I am unable to reproduce problem with "Print time :"
Can you share gcode causing the problem?
Or if it is not possible then:
slicer version
slicer settings
approximate printing time of the object
How long it took the problem to manifest itself?
any preconditions

mkbel, 2018-09-19:
Duplicate with https://github.com/prusa3d/Prusa-Firmware/issues/1151

closed 2018-09-19 by mkbel