prusa3d/Prusa-Firmware #932: MK3 Firmware 3.3.1 - Weird behavior of the filament sensor ( In a weird scenario )

bestgustavo30, 2018-07-17:
I was trying to solve a recurring problem with under extrusion and clicking extruder, then I decided to extrude manually a piece of PLA that I have,
I loaded the filament normally and started the extrusion to the end of the piece of filament, then this happened...

### **Behavior**
1. Cut Filament (for example approx 5 cm above of the extruder inlet)
2. Preheat to the temperature of your filament
3. Go to Settings > Move axis > Extruder
4. Extrude more filament of the available (for example 100 or 150 mm)
    - ### Results
       - The printer will extrude all the filament correctly until the filament sensor detects the end of the filament, then will start to retract the filament in te same velocity of the extrusion, but for the same length you tell the printer to extrude the filament ( 100 or 150 mm for this example), the printer it's not doing the process of unload of the filament.
       - then when the printer end retracting the filament for the same length of the extrusion starts the normal process of unload moving the x and y axis to the zone of unload, but to this time there is no filament to unload..
       - then the process will continue like a common load of filament.. but the printer will not resume process of manual extrusion, and some times the printer will display a random gcode in the SD card but will not start the print, (probably the last print with a event of unload of filament.
     - ### Expected results
        - The filament will extrude until the end of filament and then start the normal process of unload of the filament...
        - Optional when the filament is reloaded continue with the extrusion

I discovered the bug since the firmware version [3.2.1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.2.1) but is repicable in [3.3.0](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.3.0) and [3.3.1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.3.1) and probably previous versions

PavelSindler, 2018-07-17:
Thank you for reporting this issue.

bestgustavo30, 2018-07-17:
You're welcome! @PavelSindler :)

I need to say...
I'm not sure if the printer retract all the filament recuested by the user or only the remaining after the sensor triggers, 
I suppose that because in my experience it's seems the never ends to unload filament, Even after the gears lose grip of the filament, they still rotating for some long time in reverse, then the normal prosess of unload starts.

I hope my explanation is clear, And that the problem is easy to replicate in other machines.

Thanks, again :)

3d-gussner, 2020-07-13:
@gusdleon I think this issue has been fixed, can you please confirm and close this issue?
[stale][fixed]

gusdleon, 2020-07-15:
Can this wait a little? I'm not with my machine for probably some weeks, and I can't test if this bug is gone. Sorry :(

closed 2023-09-04 by github-actions[bot]