prusa3d/Prusa-Firmware #936: Can’t use trick to up Z axis with long press button with 3.3.1 firmware

MarcoZ76, 2018-07-17:
After upgrading the firmware 3.3.1 on my MK3 and MK2.5 i can’t use again the useful trick to elevate the Z axis with long press button.
Furthermore I can’t read the XYZ calibration detail... all result are substituted with “?”
Sorry for my bad english.

MarcoZ76, 2018-07-17:
i found the error, it's my error!!.  I compiled the firmware by myself from the repository with the last commits. Now i flash the 3.3.1 firmware already compiled from the github and all is working now. Sorry to all for this false bugs

closed 2018-07-17 by MarcoZ76

MarcoZ76, 2018-07-17:
i checked but there are many files platform.txt on Arduino folder. Which one i have to modify? I'm not good with this arduino, so could you kind show me through screenshot where and how i have to modify this file? Thanks again

Panayiotis-git, 2018-07-17:
I'm not a fan of the Arduino IDE.
I'm an eclipse guy. I'm using the very good sloeber-ide that allows you do serious Arduino programming using professional tools. Check http://eclipse.baeyens.it/
So, in eclipse I've added at the project properties, at the parameter:
"Arduino" > "Compile Options" > "append to C and C++"
the value:
-w -Os -Wl,-u,vfprintf -lprintf_flt -lm -Wl,--gc-sections