prusa3d/Prusa-Firmware #945: Mk3 FW 3.3.0-830 SD card must be re-seated to print

bobstro, 2018-07-20:
Using a mix of SD cards including original shipped with Mk3. After update to 3.3.0, status often (usually) shows "No SD card" until card is backed out slightly. Display will show progress bar, then can continue printing from card normally. Once re-seated, card will be stable until next inserted. Same cards previously worked well if card fully inserted.

workinghard, 2018-07-30:
Most likely it's a hardware issue. I experienced the same thing on my MK2 and MK3 (same sim-card slot). The sim-card slot does not work well in all seated conditions. I get used to it who to carefully insert the sim card and it's working for me.

geeksville, 2018-08-12:
Sometimes the pins on the socket aren't bent quite enough - so they have marginal connection.  You can fix that by using a small screwdriver to slightly increase the bend (the connector needs to be exposed to do this).

closed 2019-02-04 by bobstro