prusa3d/Prusa-Firmware #950: Firmware 3.3.1 only partially reads microstep settings from EEPROM on MK3

koenkooi, 2018-07-22:
I've experimented with using 32 microsteps on XY and after flashing 3.3.1 (from releases page of this github) all XY moves are doubled, so 'G1 X100' will move 20cm, not the 10cm you'd expect.

I don't need or want 32 microsteps on XY anymore, but I can only get a working machine now by doing this:

```
diff --git a/Firmware/variants/1_75mm_MK3-EINSy10a-E3Dv6full.h b/Firmware/variants/1_75mm_MK3-EINSy10a-E3Dv6full.h
index 0711d3c..e0a91f0 100644
--- a/Firmware/variants/1_75mm_MK3-EINSy10a-E3Dv6full.h
+++ b/Firmware/variants/1_75mm_MK3-EINSy10a-E3Dv6full.h
@@ -192,7 +192,7 @@
 
 #define TMC2130_FCLK 12000000       // fclk = 12MHz
 
-#define TMC2130_USTEPS_XY   16        // microstep resolution for XY axes
+#define TMC2130_USTEPS_XY   32        // microstep resolution for XY axes
```

And building my own firmware.

I'd love to reset it, but the menu where that was possible has been deleted :/

XPila, 2018-07-23:
@koenkooi  try M502 and then M500 or change resolution with M92

koenkooi, 2018-07-23:
Flashing 3.3.1 and then M502+M500 fixed it, thanks!

closed 2018-07-23 by koenkooi