prusa3d/Prusa-Firmware #968: Belt tension status check - addition to menu please

jonnyhifi, 2018-07-24:
Given  there is code written to test it ...  could it be added to as a menu item, (eg in calibration menu) as currently I believe the only way it can be measured is to run the complete “new machine” wizard which takes ages, if you’re just working on getting the belt tension sorted. This is a common thing new people with their mk3s (and presumably  other models) do once the machine is working as part of their initial “dialing stuff in” activities. To have to wait for what feels like 15/20 minutes for the machine to run the whole wizard is very frustrating whilst iterating just on belt tension - it would be great to be able to test and display the belt tension(s) alone.

KD7LRJ, 2018-07-24:
Running the "Selftest" function also tests the belt tension and on my MK3 it took took 2.5 minutes running from room temperature.

Panayiotis-git, 2018-07-24:
I agree with @jonnyhifi.

The selftest as it is currently implemented, checks in the beginning the hotend and then the belts. If you want to calibrate the belt, then after each belt regulation you have to wait for the hotend to cool and then be able to execute the selftest again.

Jacks3DPrinting, 2018-07-24:
It would be easier if we had a extra belt tension measuring option, instead of running the whole self test.

To wait between every selftest for the bed and nozzle to cool down is pretty annoying.

boelle, 2018-08-05:
does anyone even know in what file etc the selftest is "hidden" ?

simonhdma, 2018-08-23:
For me those values are bs... I have achieved 259 for Y but so much stress on the parts so I reduced it. Definitely not able to reach 240 without breaking printed parts.

twindan, 2019-10-16:
Yes, the self test takes forever to run since it has to heat up your build plate and extruder. It would be better if we could just test the belt status rather than having to go through the rest.

vintagepc, 2020-03-30:
This is fixed by #2473

PR #2473 (2020-04-20): #2472 #2356 #2484 belt test improvements
Forces high power mode for belt test.

- Fixes #2472, Fixes #2356, resolves #2484 

Edit: Also fixes #968 , closes #2547 

PFW-1104
closed 2020-04-20 by DRracer