prusa3d/Prusa-Firmware #980: Print fan error on FW 3.3.1 (Mk3 / OctoPrint 1.3.9)

shinister, 2018-07-27:
Just upgraded to FW 3.3.1, and prints are now halting during layer 2 with "Err: PRINT FAN ERROR. Sounds identical to issue #703.

100% reproducible with any gcode (via USB / OctoPrint 1.3.9), including files successfully printed prior to the FW update. 

Suggestions?

shinister, 2018-07-27:
...just to add some additional info:

Gcode were prepared using Slic3r PE 1.40, and 1.41.0-a2

Downgraded to FW 3.2.1 and 3.1.3, but still getting the print fan error on layer 2 (on the same spot).

Attached an example of the failed print gcode (prepared with 1.41.0-a2, failed on FW versions mentioned above)

[base + thin registration alpha.txt](https://github.com/prusa3d/Prusa-Firmware/files/2236963/base.%2B.thin.registration.alpha.txt)

Nickwren, 2018-07-28:
I have just upgraded and yep the next print i do im getting PRINT FAN ERROR every so often.

workinghard, 2018-07-30:
Check your print fan cable. Most likely it's broken.

shinister, 2018-07-30:
Self-test indicates the fan is working normally - is it possible to have a broken fan cable anyway?

Upon further testing (on Mk3 / FW 3.3.1) I am able to print the "batman" test object (both from gcode and from STL sliced in Slic3r PE 1.41.0-a3). When I prepare my model for printing (attached gcode sample includes up to layer 3), it fails on the second layer with PRINT FAN ERROR. Second layer perimeters go down OK, print halts during the bottom solid fill on multiple print attempts.

[1_4 test_0.2mm_PLA_MK3.txt](https://github.com/prusa3d/Prusa-Firmware/files/2243106/1_4.test_0.2mm_PLA_MK3.txt)

workinghard, 2018-07-31:
Broken cables are tend to work in some positions. You never know in which position it's failing unless you wiggle everywhere. So yes, it's most likely.
I suggest you turn your fans on and wiggle on all cables in all directions. Listen and watch when the fan is failing. Sometimes you can even locate the exact spot.

stormfarm, 2018-08-14:
I am having the same issue.  However mine is the Mk 2.5.
I can manually turn the fan on but as soon as I attempt to print or do a preheat the fan shuts down
and I get the error message.
I too have attempted the rollback with the same result.
I printed an Item prior to the upgrade and had the issue immediately after.  
Oh and I have attempted to print from Pronterface and via sd card.  Same result

Any ideas

workinghard, 2018-08-14:
Same advice for you as-well. Check the connection, check the cable. This is a mechanical issue, not firmware.

stormfarm, 2018-08-14:
Tried that prior to posting.  Just in case.  Nothing.  Since I can start the fan from the printer control and it gets shut off only when I attempt to run preheat or print something then it would appear to be firmware.

workinghard, 2018-08-14:
@stormfarm : Get in touch with a PR customer service and try to chase down the issue on your printer. With your current description it's not possible to reproduce it on a software level. Preheating function is being done millions times a day with this firmware without any issues. It's very simple and rock solid.

shinister, 2018-08-15:
@workinghard appreciate your help, i followed the steps you suggested and am not able to reproduce the fan error while arbitrarily moving the extruder to various points on the bed while wiggling the wire. 

printer never gave a fan error until this update. currently i am able to activate the fan at a variety of speed settings through the menu. 

@stormfarm consider disabling fan check on the printer settings. that did the trick for me, at least until this is resolved. zero fan errors (or other apparent hiccups) during a >50 hour stress test print with fan check off - would have expected a mechanical issue to reveal itself in this scenario.

closed 2023-08-24 by 3d-gussner