prusa3d/Prusa-Firmware #983: Feature request - manual mesh level adjustment > +- 50

jonnyhifi, 2018-07-29:
On my mk3, when trying to get uniform test first layers across the bed, I’ve ended up with left & right sides +50, front side +10, rear 0.
With a perfect middle first layer dialled in, clearly the left and right still have the nozzle too low. I can’t simply alter my live z - without compromising the centre. It’s close though - if we could have say +/- 100 maximum or even +- 150, there is no question I could tune all of them perfectly. Why not have an arbitrarily larger range than ever needed, so you don’t just run out of potential range when making these corrections ?

3d-gussner, 2018-07-31:
Hi @jonnyhifi, 
as @triplepoint said try first to get it hardware wise as accurate as possible.
But you can also modify the values and compile the firmware yourself.

in ultralcd.cpp default is 50 max is 127!
change from
`#define BED_ADJUSTMENT_UM_MAX 50`
to
`#define BED_ADJUSTMENT_UM_MAX 127
`
and
in Marlin_main.cpp
change from:
`if (fabs(offset) > 0.101f) {`
to
`if (fabs(offset) > 0.201f) {`

don't forget to change
in Configuration.h from:

```
#define FW_DEV_VERSION FW_VERSION_UNKNOWN
#define FW_REPOSITORY "Unknown"
```

to
```
#define FW_DEV_VERSION FW_VERSION_GOLD
#define FW_REPOSITORY "Prusa3D/MK3"

```
otherwise you will get LCD messages on the screen that you are using an Unkown/Beta/Alpha firmware.

I have used these in the Hyperfine Bed Leveling firmware https://github.com/3d-gussner/Prusa-Firmware/releases for quiet long time. The MK2.5 and MK3 releases aren't up2date at this moment but I am busy to build my MK3 the next days and get back to it soon.
Also started  a Wiki that may help https://github.com/3d-gussner/Prusa-Firmware/wiki

edspeds, 2018-08-27:
I've found manually leveling the bed doesn't help with this... I've added spring washers and my bed is leveled within .03mm as reported by PrusaMeshMap in Octoprint yet I still get very inconsistent first layers...  I just started trying to tweak the manual mesh level adjustment but haven't been very successful as I just can't get my front left corner dialed in on big prints...

Maybe an option to disable automatic bed leveling?  If my bed is truly within .03mm automatic leveling shouldn't be necessary, no?

@3d-gussner I may give your hyperfine bed leveling firmware a go as well.

closed 2023-09-06 by github-actions[bot]