prusa3d/Prusa-Firmware #986: MK3-3.3.1-845 does not display remaining time correct for long prints.

middelink, 2018-07-30:
Firmware 3.3.1-845 is displaying incorrect remaining time information when encountering prints longer than (it seems) 10 hours. The generated M73 S and R codes in the g-codelook fine, starting from 733/730 minutes, counting down to 0. 730 minutes matches the screenshot from Slicr PE 1.41.0-alpha3 for being roughly 12h10m.

However, the display is showing 01:05R ... and then, when that hour has passed, jumps back up.
Look like some 16 bit overflow thingy :(

[image]
[image]

quadportnick, 2018-08-01:
FWIW, I have the same issue on the MK2.5 with the same firmware build, and the same situations, so I don't think @middelink has anything unique going on. The M73 plugin for Octoprint doesn't fare any better.

middelink, 2018-08-02:
I looked at the code and it is indeed a uint16_t overflow. Line Marlin_main.cpp:8845, the statement `print_t = 100 * print_t / feedmultiply`. 730 * 100 = 73000, which is slightly over 65536 (MAX_UINT16). 
Probably making the constant 100UL might fix it, otherwise you need to add some casts to longs. Sadly, as I don't use cmake, I can't compile and therefore test.

```c
uint16_t print_time_remaining() {
        uint16_t print_t = PRINT_TIME_REMAINING_INIT;
        if (SilentModeMenu == SILENT_MODE_OFF) print_t = print_time_remaining_normal;
        else print_t = print_time_remaining_silent;
        if ((print_t != PRINT_TIME_REMAINING_INIT) && (feedmultiply != 0)) print_t = 100 * print_t / feedmultiply;
        return print_t;
}
```

middelink, 2018-08-08:
@3d-gussner Sadly no, C/C++ will use the type of the arguments to determine if it uses an integer or float division, as both arguments are uint16_t, it will do an integer division, so your examples become:

- 730min print with 100 feedmultiply: 730 / 100 would be 7 and than multiplied by 100 = 700.
- 730min print with 150 feedmultiply: 730 / 150 would be 4 and than multiplied by 100 = 400.

This is why I suggested forcing the constant 100 to unsigned LONG, so the multiplication would become long and then the division as well. As an unsigned long can hold ~4 billion, 70000 is not a problem.

3d-gussner, 2018-08-09:
Thanks for the answer...they just commit a change that may fix it https://github.com/prusa3d/Prusa-Firmware/pull/1021/commits/990a5c9119b2515d265ecb10f745371f2fee3cbb

middelink, 2018-08-09:
Yup, that would fix it too, at the cost of using 2 bytes more RAM.

Prusa-Support, 2022-05-06:
This issue is fixed by PR #1113.

Michele Moramarco
Prusa Research

closed 2022-05-06 by Prusa-Support