prusa3d/Prusa-Firmware #988: MMUv2 Feature request: Add primary/backup filament function

3d-gussner, 2018-07-31:
In the past i asked for an option to be able to define primary and backup filament extruders to the MMUv1, so i try it again for the MMUv2.

Some of us would like to finish the spools and not having rests laying around due not knowing if that is enough to finish a print.

It works great with my filament runout sensor MK2/s firmware (using optical or mechanical sensor) and the MK2.5/MK3 firmware (PAT9125 sensor).
**BUT** you have to be around to switch the filament or get noisy beeps in the middle of the night.

One of the reasons I ordered the MMUv2 is/was
- print in multi materials / colors
- hopefully be able to define primary and backup extruders/filaments to change and continue the print in the same material/color while finishing the nearly empty spool 1st and have the new one in a backup slot without doing the change manually.

Here some thoughts:

Adding next to 'Tx', 'Lx' and 'Ux' commands from the Einsy/miniRambo to the MMUv2 controller 
- 'Pyx'
  - **P**rimary
  - y defines the primary/backup group; with 5 extruders max. is 2
    - 0 = disables all groups
    - 1 = 1st group
    - 2 = 2nd group 
  - x defines the primary extruder in that group
- 'Byx'
  - **B**ackup
   - y defines the primary/backup group or common backup; with 5 extruders max is 3
     - 0 = disables all groups
     - 1 = member of 1st group
     - 2 = member of 2nd group
     - 3 = backup for all/other extruders (maybe rare but you never know)
   - x defines the backup extruder

This would be nice to be implemented in Slic3rPE as well to prevent double usage of an MMUv2 extruder once as backup and then also as normal.

- As soon the primary extruder is selected and runouts of filament the unload sequence is started.
- The MMUv2 automatically switches to the backup extruder and initiate loading of the filament. At the same time all future primary extruder commands will be mapped to the backup extruder so the gcode can stay the same and doesn't need any additional adjustments. Otherwise the next T command would switch back to the empty extruder. I am not sure that the mapping should be stored in the MMUv2 controller EEPROM or not, as normally the user should know before starting the next print which extruder needs the filament. On the other side i am curious if MMUv2 will work with the PowerPanic function, if so the mapping of the primary extruder after switching to backup should be stored in the EEPROM until it finishes the print.
- Choosing for a common backup for the all/other extruders might be strange in multi material/color prints but useful if you need lot of filament spool changes.
  - Thinking of large metal filament prints, where the length of these limits the size of the print

Example:
Printing a single extruder print but not knowing if the amount of the spool is enough you could load the 1st spool to 1st extruder and the new one to the 2nd.
Additional start-gcode command could be:
```
P11; set extruder 1 as primary of group 1
B12; set extruder 2 as backup of group 1

```

Printing a two material/color print you could make group 1 of extruder 1+2 and group 2 of extruder 3+5
Additional start-gcode command could be:
```
P11; set extruder 1 as primary of group 1
B12; set extruder 2 as backup of group 1
P23; set extruder 3 as primary of group 2
B25; set extruder 5 as backup of group 2

```

The end-gcode could be;
```
P0; disables all groups
B0; disables all groups and common backup
```

TheZeroBeast, 2018-08-07:
This is a brilliant idea that I hadn't considered before but makes so much sense, even more so no MMUv2 has 5 colors to choose from. I will give this a thumbs up, I hope in the near future this could be incorporated. 👍

danielegger1, 2018-08-07:
Thumbs up on this - great idea

And maybe an additional idea as I am finished a lot of nearly empty spools with some stupid things prints as the moment:

5 spools with a small amount of filament (could be also 5kg of filament and a huge print)

If spool 1 is empty switch to spool 2
If spool 2 is empty switch to spool 3
Etc.

So instead of one backup roll it would be four backup packs ;)

3d-gussner, 2018-08-08:
@danielegger1 with my idea you just had to define the extruders to the group, in your example:
P11;
B12;
B13;
B14;
B15;

Just needed some logic to check if the backup has filament or no while switching after it run out, if not look for the next extruder in that group with filament available.

Maybe a fall back option would be interesting too, so you could add new filament to the prior run out one and notify (Gcode or push a button combination) the MMUv2 controller to add it back to the group. Or just if all backup run out check the primary again, like a loop until it finds one extruder with filament in that group.
Also a gcode notification when it switches between filaments in a group would be great, which could be used in Octoprint to notify the user to check filament. Notification from octoprint could be scheduled so you don't wake up in the middle of the night but still could re add filament to the empty slots in the group.

brandneran, 2018-10-18:
is this request already implemented?

3d-gussner, 2018-11-20:
@brandneran Prusa implemented auto deplete, which a simple version where after running out it moves to the next mmu extruder position. So in theory you could pace a nearly empty spool in extruder 1 and at 2nd one the new spool of the same filament. 3rd again nearly empty spool and 4th new one. As far I can see it doesn't have any options other options I proposed.

3d-gussner, 2019-01-08:
Basic function is implemented, so i close it

closed 2019-01-08 by 3d-gussner