prusa3d/Prusa-Firmware #992: Unable to compile 3.3.1 (new VM, Arduino 1.6.9, modified platform.txt)

FischOderAal, 2018-07-31:
Hello everyone,

I am wondering if I'm missing something or something is missing from the description.

I downloaded 3.3.1 from the "releases" tab, made a new Windows 7 VM, downloaded Arduino 1.6.9, as per the description, added the Ultimachine boards and then edited every platform.txt I was able to find (after editing only the one in the "rambo" folder didn't work). I then copied the correct file from variants, renamed it to "Configuration_prusa.h" but still I'm unable to compile.

Here's the error message:
https://imgur.com/ZabwEQQ

FischOderAal, 2018-07-31:
So it seems like even though I downloaded 1.6.9, it still has a newer compiler than the 4.8.1 recommended.

How does that work?

3d-gussner, 2018-07-31:
@FischOderAal 
1. download https://www.arduino.cc/download_handler.php?f=/arduino-1.6.9-windows.zip
2. unzip it where you want to have it
3. create a folder 'portable' in the same folder you find arduino.exe //this makes it independent from other versions.
4. edit both platform.txt files under portable\packages and hardware\arduino as described
5. follow the instructions

**DO NOT EVER UPDATE in the board manager the 'Arduino AVR Boards' higher than 1.6.11 !!!**

Updating the Libraries works for me fine.

FischOderAal, 2018-07-31:
Oooooooh. I guess I did update the "arduino AVR boards".  Now it works! Thank you for the detailed explanation.

closed 2018-07-31 by FischOderAal

3d-gussner, 2018-07-31:
Glad to hear that it helped.
I tried it every version and with 1.6.12->higher you get more errors.

Still like more the Arduino IDE 1.8.5 as it uses less ressources but to be compatible with Prusa i use the portable version 1.6.9