prusa3d/Prusa-Firmware #1000: Opening Serial Interface stops print

NitramLegov, 2018-08-02:
Hey guys,

we´re currently trying to extend our MK2 printers at our local makerspace with a network interface. However, printing should still only be possible from the SD-Card (so that you have to physically start the print at the machine) due to security reasons.

As of this, I was trying to leverage the serial interface provided on the USB port on the MK2.

Unfortunately, the print stops as soon as I open the serial connection and the only way to resume it was to restart the print.

We are running on the latest firmware 3.1.0

For me, this looks like a bug. Can you confirm that?

Thanks

NitramLegov, 2018-08-02:
> As soon you establish a serial communication the printer will reset. You should initiate the communication before a print. I use Octcoprint on a C.H.I.P. and ESP3D to communicate with my printers and have to make sure that none restarts while it is printing.

Yeah, that's the behaviour I saw. Is this done purposely? Is it a hardware behaviour or is it a firmware "feature"?
In my case, it is not the desired behaviour. Especially when I am printing from the SD card and not via the serial interface, so I thought it might be a bug.

bubnikv, 2018-08-02:
It is a feature of the Arduino Mega. The Einsy Rambo matches the behavior
in this respect.

boelle, 2018-08-04:
there is a small brigde somewhere you can cut so it does not do this

but then you cant flash firmware... 

[image]

what i did on a mega2560 board was to solder on 2 wires and have a switch so that i can open and close the connection when i need it

Area5142, 2018-08-11:
It looks like it would be possible to solder a two pin header for a jumper on the two pads?

geeksville, 2018-08-18:
My other 3D printer - which uses a classic Rambo board and Arduino MEGA does not have this behavior.  i.e. connecting to the printer with Octoprint does _not_ force a reboot of the MCU.  This other printer also uses Marlin.

I've used 2560's on boards I've designed and I'm not aware of anything in the hardware that forces a reboot on serial connection.  I think this is merely a bug in the Prusa fork of Marlin.

maximlevitsky, 2019-07-30:
This is not possible to fix in the main firmware of the printer, unless you want to flash the firmware of the AtMega32u2 chip which is used for usb<->serial conversion.

Basically what happens is that when the PC opens the virtual usb serial port, it toggles the virtual DTR wire, which causes the firmware in AtMega32u2, toggle the reset pin of the main MCU (AtMega2560)

This is arduino lefover, because by convention, the bootloader of the main MCU checks if it runs because it was powered up or because it was rebooted. In later case, it waits a bit on the serial for firmware upload commands. This was done so, it is easy to update the firmware on MCU. You just connect over the serial port, that reboots it and vola, it accepts the firmware flash instructions.
On the other hand when the device is powered up, it goes straight into the user application (sketch in the arduino terms).

You can partially mitigate the issue by having this udev rule on your system:

SUBSYSTEM=="tty", ATTRS{idVendor}=="2c99", KERNEL=="ttyACM*", RUN+="/usr/bin/stty -F /dev/%k -hupcl"

This will cause the printer to reboot once when you connect it to the PC (since the stty command run here, opens the serial port, but then this command disables the DTR toggling, and thus on next port open the printer doesn't reboot, and that as long as you don't disconnect the printer from the PC)
The printer doesn't even disconnect when it is powered down, since the USB converter chip is powered by the PC power.

Going even further, it is possible to find the place in the Linux kernel that you use on the host that does the DTR toggling and disable it by default,

Or you can change the AtMega32U4 firmware to stop the RESET toggling, but that will make it harder to flash the firmware. You would have to reset the printer by hand (with the reset button for example, and then in the window of few seconds star the firmware upload).

beegmouse, 2019-08-08:
Does the bridge mentioned above exist on the einsy Virtual DTR line?

Can we wire in a jumper/button we could bridge when we want to do a firmware update?

from what you are saying a hardware solution is the better way forward. Having the user hold a button down when wanting to update firmware.

leptun, 2020-01-26:
I know this is an old issue, but there is a solution to this that doesn't require any hardware change.
All you have to do is send these commands over regular serial connection to the printer (putty is recommended):
`;C32u2_RMD` - Disable reset on connect
`;C32u2_RME` - Enable reset on connect (default)
`;C2560_RES` - Resets the printer.

Also, would you mind closing this issue?

closed 2023-09-06 by github-actions[bot]