prusa3d/Prusa-Firmware #1008: M73 Time Remaining not working when print over 12h

TheZeroBeast, 2018-08-05:
Hello,
I'm running MK3 on 3.3.1 and Slic3r 1.41.0Alpha3 and when a print is over 12h I get the excess displayed counting down to 12h remaining which then starts counting down the final 12h.

The attached gcode has M73 commands correct showing 18h in number of minutes so it appears to not display correct on MK3 at present.

[FEDAIM-Case-Front_0.2mm_ABS_MK3.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/2260033/FEDAIM-Case-Front_0.2mm_ABS_MK3.gcode.zip)

Thanks,
Rob

3d-gussner, 2018-08-05:
Maybe there is an issue with values above 1024 as 18h x 60m is 1080...would be interesting to generate a gcode running for 17h (under 1024min) to prove it.

TheZeroBeast, 2018-08-05:
[image]
Here is that gcode 3h into the 18h print so might not be a variable limit as the dial cause. 🤔

RacingHell, 2018-08-07:
Can confirm this. A 14h40m print starts counting at roundabout 3 hours remaining.

3d-gussner, 2018-08-08:
For me it looks like 12h is the point of turn, as @RacingHell has a 14h40m print starts counting opposite direction around 12h into the print ...14h40m - ~3h = ~12h

TheZeroBeast, 2018-08-09:
I’d say this commit will solve the issue. Will need to test.

https://github.com/prusa3d/Prusa-Firmware/commit/990a5c9119b2515d265ecb10f745371f2fee3cbb

middelink, 2018-08-09:
This is dupe of #986

Prusa-Support, 2022-05-06:
This issue will be closed as a duplicate of issue #986.

Michele Moramarco
Prusa Research

closed 2022-05-06 by Prusa-Support