prusa3d/Prusa-Firmware #1009: Filament Sensor Detection Option Discussion/Ideas

dot-bob, 2018-08-05:
So the filament sensor can detect filament presence and or movement/jam.  It also has the issue of occasionally false triggering an error with certain types of filament.  Is the typical error when false triggering a filament presence error or a movement error?

If the false positive is due to movement/jam detection failing could we get better sensor reliability with troublesome filaments if we can disable movement detection and leave presence detection enabled?

clearbucketLabs, 2018-08-09:
it's a hardware/sensor issue.  there's no movement detection, just presence in firmware.   you could use the sensor on a bearing/wheel that the filament moves and that would work (indirectly).

RetireeJay46, 2018-08-29:
Are you sure it's just a presence detector?  The active device on the filament sensor board is a laser sensor package used in optical mice, which definitely do sense motion.

clearbucketLabs, 2018-08-29:
Yes but the firmware doesn't track the movement.   my guess is it's just a periodic check if it's present... if it did motion the rate would have to be higher than the CPU could probably handle at this point as they already have motors and two fans to monitor in addition to the usual hot and and bed too.  if it calculated motion it could use the feedback to adjust the extrusion rate but it does not. :(

mgx0, 2018-09-01:
Since the sensor is not reliable across all filaments, I would propose to add third option to the Filament Sensor: On, Off and Load (example). This would disable the filament sensor when printing and enable it when not printing. Filament Autoload is useful feature, but it's annoying to either have autoload enabled and having fake alerts during printing, or No alerts furing printing and Autoload off. 

Simply, just let us to use the sensor Autoload only. In current state it's causing more problems than it serves it's purpose.

clearbucketLabs, 2018-09-01:
i haven't had a whole lot of problems with it so far, even on shiny black petg.

jltx1, 2018-09-09:
Here is my solution: https://www.thingiverse.com/thing:3091625

clearbucketLabs, 2018-09-10:
@jltx1 saw that, thanks :)  apparently this is how it's supposed to be used anyway...

closed 2023-09-04 by github-actions[bot]