prusa3d/Prusa-Firmware #1017: LCD Displaying "Sleep..." During USB Printing

TheZeroBeast, 2018-08-07:
Hello,
I'm running MK3 on 3.3.1 and Slic3r 1.41.0Beta.

During USB printing(SD Card printing may do the same) the LCD bottom info displays "Sleep..." as per screenshot.

Note: as per FW 3.3.0 release notes M86 S0 & M86 S1800 have been added to my Octoinstance, not sure if this would be contributing at all.

[image]

Thanks,
Rob

ptfly, 2019-10-27:
Same thing is happening to me on the latest FW on MK3S. I'm using Gcode to trigger Einsy pin and take a photo on a GoPro through esp8266 chip via `M42 S255 P73` and then using `G4 P200`. Once the delay is used, the message isn't cleared from the screen.

ptfly, 2019-10-31:
v3.8.1 - same thing, not fixed :( May be this issue is too deep in the timeline!

3d-gussner, 2020-07-13:
I am reviewing issues and think this one isn't fixed.
What do you expect after G4 is timed out?
My guess that the "Sleep ..." message disappears and the "normal" Info Screen is shown again.

closed 2023-09-06 by github-actions[bot]