prusa3d/Prusa-Firmware #1034: Feature request: pre-heat nozzle only based on set filaments

alexwhittemore, 2018-08-13:
This would make the filament load workflow much easier. Given that this is the most common manual interaction I have with the printer aside from printing from SD (which is already pretty fast), it feels like there's much room for improvement.

**Current Workflow:**
* Turn on machine
* Navigate to Settings (click+5 steps +click)
* Navigate to Temperature (1 step + click)
* Navigate to Nozzle (1 step + click)
* Set temp (190 to 230 steps, paying attention to use the right number, + click)
* Navigate back to menu (1 step + 3 clicks)
* Wait for temp to come up
* Navigate to Load Filament (click 3 steps click)
* Tend to the load
* (often) repeat Load Filament rather than clicking "No, not clear," since the faster initial extrusion often clears out old overcooked plastic better.
* Reset the printer to turn off the nozzle heater

**Desired Workflow:**
* Turn on machine
* Navigate to "Load Filament" (click, one step, click - same position as current "preheat")
* Navigate to "heat nozzle" (click, one step, click)
* Navigate to (PLA/PETG/Nylon/etc) (one, two, three... steps, click)

At this point, perhaps the flow could be integrated or perhaps you could be thrown right back to the main info screen for more flexibility. I'll assume the latter.
* Wait for nozzle to come up
* Navigate to Load Filament (click, one step, click)
* Navigate to "feed" (two steps, click)
* Tend to the load
* Respond to modfied question "Keep feeding fast, keep feeding slow, done"
* Reset printer to turn off nozzle heater.

Note that I've replaced preheat - I think probably you bump that down one item. I personally don't ever use preheat, since I'm still on a MK2s and preheating throws off the PINDA probe. Instead, my custom gcode does something like:

* nozzle to 170
* raise away from bed
* bed to first layer temp
* mesh bed level
* home
* nozzle to first layer temp
* prime and print

I'd be willing to bet that most people, between some kind of custom start like that and octoprint, don't bother with "preheat" either, so it'd probably be safe to move it one item space to make filament loading easier.

dalgibbard, 2018-08-21:
True enough - I only ever use preheat to do something hot-end related (change filament etc); and heating the bed is actually inconvenient :)

simonhdma, 2018-08-23:
Thumbs up from me! 👍 :)

cwei, 2018-08-23:
From me too 👍

alexwhittemore, 2018-08-23:
Thanks for the love!

noxhirsch, 2018-08-25:
I like it, too 👍

alexwhittemore, 2018-08-25:
Huh, not a bad idea! I don’t think that obviates the request here because it’s still more cumbersome than need be, but that’s certainly an improvement on digging through menus, thanks!

Of course, another easy way to do it is to just use Octoprint, but it happens to be that I don’t always have that set up.

Prusa-Support, 2022-05-19:
Duplicate of #422

closed 2023-09-06 by github-actions[bot]