prusa3d/Prusa-Firmware #1037: Resume printing after a pause.

Thruxton78, 2018-08-14:
I am currently using a Prusa i3 Mk3, and sometimes the "pause" function during a print.
But as some material keeps on flowing from the nozzle during the pause, when you resume the printing process, a small amount of plastic can form a non desired feature at the restart point ...
Would it be possible to suggest a purge of the nozzle on the side of the bed before restarting the printing in the next upgrade of the firmware ?

austinvojta, 2018-09-05:
Not a bad idea, but I can suggest that in the meantime you use the "Tune" -> "Change Filament" option to pause your print instead. This way it will think you're loading a new filament and do the purge before it resumes.

Thruxton78, 2018-09-05:
Hi,
Ok thank you for this trick. I'll manage with that in the meantime.

closed 2023-09-05 by github-actions[bot]