prusa3d/Prusa-Firmware #1039: Hotend Temp wrong when starting print

DataCrypt, 2018-08-14:
On my MK3 (latest firmware) I had filament loaded (filament sensor disabled) and selected "Print from SD card".  It said "Bed heating..." and the bed went up to 60 and then starting printing.  Yet the Nozzle temp was still showing 0.  I had to stop print immediately and go into settings, Temperature, and Nozzle to preheat it.  Shouldn't this have preheated along with bed?  If it did, the display showed 0.

GurliGebis, 2018-08-16:
Can you paste the first 20 lines of the G code file you where printing?

DataCrypt, 2018-08-17:
@GurliGebis - here is the first part of the gcode file.

`; generated by Slic3r Prusa Edition 1.40.1+win64 on 2018-08-12 at 11:10:28
; 
; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M107
M115 U3.3.1 ; tell printer latest fw version
M201 X1000 Y1000 Z1000 E9000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 S1250 T1250 ; sets acceleration (S) and retract acceleration (T)
M205 X8 Y8 Z0.4 E1.5 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M83  ; extruder relative mode
M104 S208 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S208 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
G92 E0.0
M221 S95
M900 K30; Filament gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.2`

GurliGebis, 2018-08-20:
Okay, the gcode looks like the normal start script, so that should work.

Prusa-Support, 2022-06-06:
Hello! I am absolutely sorry for not intervening on this thread before. Was the issue solved in the meantime? In case it wasn't, was the 0 temperature the "measured temperature" or the "target temperature"? It could have been a HW issue - in case there are still similar issues, can you please let me know via email? info@prusa3d.com mentioning me. 

Alessandro Pantaleo
Prusa Research

closed 2022-06-06 by Prusa-Support