prusa3d/Prusa-Firmware #1040: Ghost switching of print speed

Skywatchr, 2018-08-15:
Using the firmware from last night without any modification or customisation I encountered a strange effect today when entering the menu to change some values in the Tune submenu.
After changing the for example the nozzle temperature and returning to the info screen, the printer applies the changes and in addition switches to 345% print speed
I can reproduce this with different settings, sometimes even without changing anything and just returning to the info screen.
The problem persists after recompiling and reflashing the firmware 

configuration.h reports:
firmware 3.4.0-RC1
commit 1170

https://youtu.be/HeDU83OcUZY

XPila, 2018-08-15:
@uepsie we will try to reproduce, please use original firmware https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.4.0-RC1 and use Slicer to flash the firmware, it can be problem in your build system.

uepsie, 2018-08-15:
@XPila I compiled it on a System running Windows 10 and a clean Debian Stretch with nothing additional but the Arduino 1.6.8 IDE on it.
All firmware files have been flashed using Slic3r. The problem is still there.
I went back to 3.3.0-880 and it disappeared. 
Then I cloned the -b MK3 from https://github.com/prusa3d/Prusa-Firmware.git, compiled and flashed it and the strange behaviour is back again :-)

XPila, 2018-08-16:
@uepsie we reproduced same behavior, it will be fixed asap.

leefla, 2018-08-17:
@XPila do you have any idea what commit may have introduced this ?
If so I might roll-back my fork to before that commit. Just thought you might know of a CL that touched that area, more easily than me searching through all recent commits.

PavelSindler, 2018-08-26:
Fixed in PR https://github.com/prusa3d/Prusa-Firmware/pull/1053

closed 2018-08-26 by PavelSindler