prusa3d/Prusa-Firmware #1043: after fil sensor response is poor message, disabled sensor and then the printer reset! (trans petg)

mpkottawa, 2018-08-15:
this was my first print after updating to this firmware. i was using octoprint and had a transparent petg print at about 80%. came up to the printer with it beeping.  after unloading, and when it came up, i clicked disable sensor, then i noticed it showed"fillament loaded correctly" screen was barely legable(very dim)  and right after that, the printer just reset.

michalzielanski, 2018-08-18:
The same with the M600 command. Filament unload, load and firmware reset.

GurliGebis, 2018-08-20:
Does it also happen if you copy the gcode file to the SD card and run it from there?

Just to rule out Octoprint seeing it as the printer hanging, and then trying to reconnect, and by doing that resetting it.

michalzielanski, 2018-08-23:
I did not check the print from the SD card, but the same version of Octoprint plus FW 3.3.1 works properly.

Phydam, 2018-08-23:
I ran a test print on that FW3.3.1 from Octoprint and SD card without incident. I updated to 3.4.0-RC1 and printing from the SD card worked alright but it reset on filament loading when printing from Octoprint.

GurliGebis, 2018-08-27:
Sounds like Octoprint thinks the printer is timing out, and resets it.
Maybe the firmware is not telling Octoprint that it is busy.

austinvojta, 2018-09-05:
I had an issue that I think (?) was related to this recently on 3.4.1 RC2. My printer was left beeping for about 5 hours when it hit an M600 command in the gcode. The hotend hit the time limit and cooled down, then I came back and preheated the nozzle to unload and then load the next filament color. After doing that, the printer spazzed out and started moving diagonally across the bed while lowering the head (moving Z down) straight into my partially printed model. I turned it off immediately and had to restart the print.

talldonkey, 2018-09-06:
Check out this workaround:  https://github.com/foosel/OctoPrint/issues/2026 . GurliGebis, you're headed in the right direction.

closed 2023-09-05 by github-actions[bot]