prusa3d/Prusa-Firmware #1047: 3.4.0-RC1 Firmware for Prusa i3 MK3 filament sensor on or off causes speed to move over 300

TimothyGold, 2018-08-20:
I noticed that after loading 3.4.0-RC1 Firmware for Prusa i3 MK3 and playing with the new improvements in the filament sensor after finding that the quality is low and it asked to disable the filament sensor and choosing yes the speed of the print went from 100% to over 350%.

mkbel, 2018-08-20:
I am unable to reproduce problem.

TimothyGold, 2018-08-20:
When I experienced this it happened twice in the same print.  I was printing from Octoprint. Slicer is S3D.
1. - Was in the middle of a autoload with sensor enabled.
2. - Started print via octoprint before autoload prime was complete.  
3. - somewhere in the process it stated that the sensor quality was low and asked to disable it.
4. - I said yes.  
5. - When the print finally started it started off 100% but then jumped to 350%.  I manually turned it back down to 100%.  I continued to play around with the sensor to enable it again then ended up disabling it again.  It then did it again.  I don't know for sure what is triggering it but I would assume it was the filament enabling/disabling.  

I reverted back to stable firmware.

dalgibbard, 2018-08-21:
Possibly related to #1040

cosmith71, 2018-08-26:
I believe I had a similar issue with the MK2.5.  I began having Y axis layer shifts and noticed that the printer appeared to be moving much faster than usual, even though the LCD said 100%.

PavelSindler, 2018-08-26:
Fixed in PR https://github.com/prusa3d/Prusa-Firmware/pull/1053

closed 2018-08-26 by PavelSindler

sadistiko, 2018-09-19:
I also have similar problem. After upgrading to Firmware 3.4.0 (last night) on my MK3 I am asked to disable the filament sensor after every color change (M600 code). Also I think that now temperature goes to 215°C slower than before. I tried to revert to last used Firmware (3.3.1) but Slic3r refused to do that. I'm in middle of some big job and I don't have time to play with settings.

mkbel, 2018-09-21:
Thank you for reporting, but this issue is very different from your three issues and is already closed. Next time please create new issue (if it already doesn't exist). Create separate issues for separate problems. Create issue related to Slic3r on Slic3r github https://github.com/prusa3d/Slic3r/ . To your three problems:

1. Filament sensor: The check of quality response from filament sensor is new in this release. Check if your filament sensor is not covered by dust or thin filament strings. If this is not the case, you can create issue for that.

2. Temperature: It is very improbable, as we didn't change nozzle temperature handling. You can try any of following, and please create separate issue with your findings:
a) Measure with stopwatch temperature rise from some low temperature to some higher temperature e.g. from 50 deg. C to 200 deg. C. Downgrade firmware and measure again.
b) Measure with stopwatch temperature rise from some low temperature to some higher temperature e.g. from 50 deg. C to 200 deg. C. Run PID calibration and than measure again.
c) Measure with stopwatch temperature rise from some low temperature to some higher temperature e.g. from 50 deg. C to 200 deg. C. Do the factory reset and than measure again.

3. Slic3r: I asked Slic3r team and they told me that there is no firmware version check in Slc3r. If you are encountering still this, please report it with all details on https://github.com/prusa3d/Slic3r/
If you are on Linux (reproduced on Debian), there is known problem with Modem Manager. If you connect or power up the printer, first about 30 seconds printer is not available for Slic3r, as is enumerated by Modem Manager. This problem is reported and solved by Debian community.

mkbel, 2018-09-21:
1. To filament sensor: We have found, that even we throughly tested measurement of filament sensor response quality on particular filament during ordinary load. The load during color change (M600 code) behaves worse and we will temporary disable this in next release.

bubnikv, 2018-09-21:
@sadistiko 
> I tried to revert to last used Firmware (3.3.1) but Slic3r refused to do that.

What happened exactly? If you think there is an issue with Slic3r's firmware updater feature, then it would be best to enter an issue into Slic3r's github issue tracker:
https://github.com/prusa3d/slic3r/issues