prusa3d/Prusa-Firmware #1048: How to Properly Compile Firmware 3.4.0-rc1

ojrik, 2018-08-20:
Greetings.
I would like to advise about compiling in ARDUINO firmware 3.4.0-rc1
I use:
Windows 10 For v1803.
Arduino 1.6.9 for MK3
Arduino 1.6.5 for MK2.5
The problem is this:
in the statistics menu, XYZ cal. details all values, Move AXIS X, Y, Z, is displayed instead of values (?)
support does not display ip address for FLASHAIR.

Would you guys know what I'm doing wrong?
How to insert languages into a compiled HEX file.
Thanks for help.

cimoalpacino, 2018-08-20:
I don't think you've read the readme file.
There are instructions how to properly compile it. I've spent some time myself before i realized it was all written in readme.

modify platform.txt to enable float printf support: "compiler.c.elf.flags=-w -Os -Wl,-u,vfprintf -lprintf_flt -lm -Wl,--gc-sections"

cimoalpacino, 2018-08-20:
Also, please close this thread.

closed 2018-08-20 by ojrik