prusa3d/Prusa-Firmware #1049: Firmware retraction bug

Zingabopp, 2018-08-20:
Firmware Version: 3.3.1

When you do a G10 (retract) command it sets the boolean retracted[active_extruder] to true, probably so you can't retract over and over again. The problem is that if you have a G10 retract in your end gcode or Octoprint's "Cancel Print" gcode script the retracted variable doesn't reset. So on your next print it homes and does the purge line, fails to retract because 'retracted' is still true, and then unretracts when it starts the perimeter. If you have Z-lift enabled your nozzle height is now that amount lower than it should be.

It seems like an easy fix would be to set 'retracted' to false whenever the printer homes or does mesh bed leveling.

Zingabopp, 2018-08-21:
I don't think it would make a difference to split them up. From what I understand in the firmware, the only thing 'retracted' affects is whether or not the printer allows a retract/unretract to happen. After homing, the purge line will take care of any previous retracts that didn't get unretracted.

tcm0116, 2018-08-21:
This was addressed in Marlin by https://github.com/MarlinFirmware/Marlin/pull/9384 and https://github.com/MarlinFirmware/Marlin/pull/9390.

closed 2023-08-24 by 3d-gussner