prusa3d/Prusa-Firmware #1058: EXTRUDER SKIPPING after MK2 to MK2.5 upgrade and new firmware

plankton99, 2018-08-23:
I have literally 50 solid hours trying to fix it, Filip from Prusa said that the latest Firmware lowered the current to the extruder to cool down the extruder from heating up too much, so he said try a firmware pre 3.20 that had a higher current for the mK2.5. it worked for 5 min then started skipping again, I AM SO FRUSTRATED. Any solutions so far from this group?  please help

kelna2, 2018-09-04:
I bought my MK2 almost 2 years ago, and have had close to 0 issues with it. I updated the firmware a couple weeks ago and I am experiencing the same issues plankton99 is having.

I can manually extrude PLA just fine so it's not clogged. It seems to happen 10-15 mins into a print.

Here are all the things I've tried:
Checked filament to make sure it wasn't twisted/tangled
Adjusting tension of the idler tighter/looser
Cleaning the teeth of the extruder gear with a wire brush
Bought a brand new E3D v6 Hotend and replaced my old one
Lowered print speed in Print Settings > Speed 
Raised/Lowered print temperature for first layer / rest of print
Reduced/Disabled retraction in Printer Settings > Extruder 1 > Retraction > Length = 0

I am currently printing a model and so far so good. It seems like disabling retraction, and switching to a different PLA filament is working. I'll follow up in the morning with my results.

If that doesn't work I'm going to explore adjusting the current for the extruder stepper motor using the M907 gcode command. Wish me luck for both of us plankton99!

kelna2, 2018-09-04:
Ok disabling the retraction helped, but I could still see layers that weren't extruding enough PLA.

I switched my filament back to the old one I was having issues with and I reset all of my settings to the defaults for PLA in Slic3r 

I increased the current of the extruder stepper motor from 500 to 625. I originally did 700 but the stepper motor got a little warm, not hot, just too warm for my liking, 625 is extruding nicely!

To adjust the current to your stepper motors add this line of gcode to the printer tab in Slic3r. It should go right above "; Start G-Code sequence START"

M907 X540 Z830 E625; set current of motors XY Z E

Here is all of my custom gcode from Slic3r > Printer Settings > Custom G-code for reference:

M115 U3.1.0 ; tell printer latest fw version
M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 S1500 T1500 ; sets acceleration (S) and retract acceleration (T)
M205 X10 Y10 Z0.2 E2.5 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
**M907 X540 Z830 E625; set current of motors XY Z E**
; Start G-Code sequence START
T?
M104 S[first_layer_temperature]
M140 S[first_layer_bed_temperature]
M109 S[first_layer_temperature]
M190 S[first_layer_bed_temperature]
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
G28 W
G80
G92 E0.0
M203 E100
M92 E140
G1 Z0.250 F7200.000
G1 X50.0 E80.0  F1000.0
G1 X160.0 E20.0  F1000.0
G1 Z0.200 F7200.000
G1 X220.0 E13 F1000.0
G1 X240.0 E0 F1000.0
G1 E-4 F1000.0
G92 E0.0

plankton99, 2018-09-04:
My extruder skipping was with the Prusa PLA (gray), I tried Hatchbox PETG and works fine.  This extruder skipping is so frustrating.

EddyMI3d, 2018-09-06:
I assume you all checked this: I also had skipping with the extruder and the reason was, that the idler was tighten way too much.

devonbleak, 2018-11-03:
My MK3 was having similar issues.  After I ripped apart the entire extruder assembly and ripped out my hair because I couldn't find anything physically wrong, downgrading from 3.4.1 to 3.2.1 got me through the 7h22m print that used to jam around 2h.  Extruder clicking/skipping was reduced drastically and all layers filled in beautifully, though it was still clicking occasionally.

Krazok, 2020-08-09:
So, I see that all of these posts are very old. However, after upgrading to the latest firmware from Prusa, I am experiencing the same issues on my MK3S. I have done a cold pull, verified alignment of the gears, adjusted the idler gear tension, and completely disassembled and reassembled.

I have not tried adjusting the extruder stepper current, but that will be my next step. I'm curious if anyone else has had this crop up since the release of 3.9.0? I did not have any issues with the previous version.

3d-gussner, 2020-08-09:
As this is quite old issue can you please provide some more details
Printer
FW version
M503 results

One option is to factory reset all data if you don't mind to loose your statistics. This will erase the eeprom and fill in with default values.

Krazok, 2020-08-09:
Yes my printer is a Prusa MK3S. 
Firmware is 3.9.0. 
I"m not sure what an M503 is. 
Aside from some minor tweaks in the PrusaSlicr related to first layer size and detail, everything is still set at default.

EddyMI3d, 2020-08-09:
M503
https://marlinfw.org/docs/gcode/M503.html

3d-gussner, 2020-08-09:
Connect your printer via USB and start Pronterface (is a part of the Prusa Driver package).
In Pronterface choose the com port to your printer and hit connect.
Your printer should restart and you should see some terminal output in Pronterface.
Type M503 and confirm with enter to send the gcode command to the printer. 
The output should list the most important settings of the printer.

Prusa-Support, 2022-05-19:
This thread could be related to issue #765.
Also, this thread has been silent for years.
For these reasons, this issue will be closed.

Probably the last update (from Aug 2020) doesn't concern the main topic of this discussion, because the MK3S printer has a different design and the firmware was severely changed over time. This "new" issue may actually be a consequence of heat-creep or under-extrusion, which are topics also quickly covered in the closing comment of the same issue #765.

Please check the issue mentioned above to learn more about the conclusion of this case.
Comment below or feel free to open a new issue if you feel this issue deserves a separate discussion.

Michele Moramarco
Prusa Research