prusa3d/Prusa-Firmware #1064: Feature request - Report time remaining on M27

petrzelka, 2018-08-24:
Now that the firmware supports M73 (set time remaining) which is generated into gcode file by slic3r, it would be great if this precise status can be reported within M27 (Report SD print status).

I'm usually printing from SD and using octoprint to monitor priting status and for camera streaming and I see it sends M27 and calculates progress just from printing bytes, which is usually quite off.

If CardReader::getStatus() would be sending also time remaining and percent progress (got from M73 codes), octoprint could then use it to show exact progress in its web console. 

If you think this is not a bad idea but you don't have resources to implement it, I could take a look into it, just let me know.

Thanks!

petrzelka, 2018-08-24:
Oops, I see now that it's already implemented (yet I can't find it in the codebase)
This is from octoprint terminal:

Send: M27
Recv: NORMAL MODE: Percent done: 1; print time remaining in mins: 182
Recv: SILENT MODE: Percent done: 1; print time remaining in mins: 197
Recv: test_0.15mm_PLA_MK2.5.gcode
Recv: SD printing byte 83855/9057980
Recv: 00:04
Recv: ok

I didn't see that before probably because of https://github.com/prusa3d/Slic3r/issues/1161

Anyways, this can be closed, sorry for bothering.

P.S.: I really can't find it in the code (I have MK2.5 v3.3.1)
P.P.S: I have no idea who Gina is.

closed 2018-08-24 by petrzelka

petrzelka, 2018-08-24:
I get it now.
Recv: NORMAL MODE: Percent done: 1; print time remaining in mins: 182
Recv: SILENT MODE: Percent done: 1; print time remaining in mins: 197

those messages were not related to Send: M27 that was just coincidence.
printer writes those to console whenever it parses M73 in gcode.
this may be enough for octoprint to parse it, however it would be great to send only the message that is related to current mode (NORMAL or SILENT not both)

JohnOCFII, 2018-08-24:
> P.P.S: I have no idea who Gina is.

Gina is the developer of OctoPrint.