prusa3d/Prusa-Firmware #1096: Layer shift after power failure recovery

stanzlavos, 2018-09-01:
I have a Prusa i3 Mk3 kit.

I see the problem almost every time a print is recovered after a power failure. There is usually a half mm X axis layer shift. 

Is there any known fix for the issue?

otaviojr, 2021-07-06:
I have this problem ater a power failure today… half mm on X axis. 

I know this is an old thread, but, anyone find a way to solve it?

regards

Prusa-Support, 2022-06-24:
This issue could be closed because stale, but I'm wondering if the problem persists and if you are still able to reproduce it. 
Videos and pictures can help understand and evaluate the problem.

Because the conversation has been mostly silent so far, I guess this is not a popular concern among the users in the community. Similar concerns were observed in different scenarios though (e.g. issue #1359).

This may actually be the case that the homing on the X-axis was not performed accurately due to external factors or HW/assembly related inconsistencies.
For example, the cable wrap behind the Extruder block could be thicker than the expected; the cable wrap could be tenser than the expected; the belt may not be tight enough; the Einsy board case may be bulged or not completely closed; lubricant may have a build-up at the ends of the axis, the cable wrap was accidentally grabbed because saggy...
These are only a few examples of causes for the X-axis to bounce and home softly, or randomly home in the wrong position. In some of these cases, the homing position could be affected to a certain extent, causing inconsistent homing and minor layer-shifts when resuming a print after an XY-homing.

As long as the issue is not reproducible on a larger scale, there is no confirmed concern on the firmware side.
There is a chance the problem can be resolved with a proper troubleshooting session. Please consider reaching out the Technical Support before posting new similar problems.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

closed 2023-09-06 by github-actions[bot]