prusa3d/Prusa-Firmware #1097: 3.3.1 Z under 20 change filiment error while printing

max05210238, 2018-09-01:
When I try to change filiment while printing under Z 20, I pause the print, unload filiment, load filiment, but because Z is under 20, it raises up Z the current height +20 which is right, but when I resume the print, it didn't lower back to the original current print height, I not sure if this is a bug or an improper way to do this.

austinvojta, 2018-09-05:
Are you using the "change filament" feature, or simply pausing and unloading/loading it manually? In the middle of a print you can go to Tune -> Change Filament and it will operate correctly. I do understand the issue you're saying about it raising to load a new filament and then not re-lowering though, but meantime maybe that will help you out.

max05210238, 2018-09-05:
Thank you, and yes, after I post this, I find out I can use the Tune function to change filiment, didn’t try it yet, but still like to address this issue, maybe they can fix this in the next firmware build? Because the most direct thought is pause prints and change filiment, right? Or just me? LOL

closed 2023-09-05 by github-actions[bot]