prusa3d/Prusa-Firmware #1100: 3.4.0 RC2 bug

jltx1, 2018-09-03:
There is s bug where you can see the firmware misinterpreted the gcode and collapsed the bottom perimeter and moved the side perimeter across the middle. I confirmed that the gcode out of slic3r is correct by loading into a third party gcode viewer which matches slic3r.  So this appears to be a firmware bug. 

[image]

jltx1, 2018-09-03:
I reverted to v3.3.1 and the exact same gcode file is printing correctly now.

PavelSindler, 2018-09-04:
Was it prited from SD card? Did that happen more than once? Can you please provide gcode file?

jltx1, 2018-09-04:
Yes, from SD card.  This was 2 perimeters and multiple passes, so it was consistently making the error.  I will try to dig up the gcode this evening.  

-- Jon

PavelSindler, 2018-09-05:
So did you find it? It would be very helpful.

jltx1, 2018-09-05:
Yes, but won't let me attach.  I renamed it .png to fool to file loader.  Rename to .gcode.  Good luck.

[image]

PavelSindler, 2018-09-05:
Thank you! Testing...

jltx1, 2018-09-05:
Not sure if this is useful, but this is the partial hash it showed when I did the git pull:  Updating 8cea3a7..1774378

I used Arduino 1.85 to compile.

PavelSindler, 2018-09-05:
Ok, we can't reproduce the issue with gcode you provided. It seems that it will be ralted to fw version which you are using or eeprom state. Can you try 3.4.0-RC2 firmware version?

jltx1, 2018-09-05:
I was using RC2, but maybe compile got corrupted.  I will try again.  Thanks for checking.  

-- Jon

jltx1, 2018-09-05:
I re-pulled 3.4.0 RC2 and recompiled.  The binary is not the same size in bytes, so something is slightly different.  I flashed and am unable to reproduce this bug.  So either someone fixed it on HEAD or I somehow got a corrupt build.

closed 2018-09-05 by jltx1