prusa3d/Prusa-Firmware #1103: 3.4.0 RC2 time remaining

ironjoe13, 2018-09-05:
Started a print multiple times, with Slic3r stating it would be 14H and some minutes. After starting print, time remaining on LCD says 3H 25M, runs down to zero, then restarted remaining time at 11H.
Same Gcode reports correctly in 3.3.1

PavelSindler, 2018-09-05:
Thank you for reporting this. Can you please provide gcode file?

ironjoe13, 2018-09-05:
[city.zip](https://github.com/prusa3d/Prusa-Firmware/files/2354133/city.zip)

PavelSindler, 2018-09-06:
Working on it! Thanks.

3d-gussner, 2018-09-06:
Shouldn't https://github.com/prusa3d/Prusa-Firmware/commit/990a5c9119b2515d265ecb10f745371f2fee3cbb have fixed this issue?
See also https://github.com/prusa3d/Prusa-Firmware/issues/986

ironjoe13, 2018-09-06:
For me, 3.3.1 displayed the correct time remaining with Slic3r PE 1.41.0 on everything including a 42 hr print.

PavelSindler, 2018-09-07:
PR https://github.com/prusa3d/Prusa-Firmware/pull/1113 fixes this issue.

closed 2018-09-07 by PavelSindler