prusa3d/Prusa-Firmware #1106: MK2.5  3.3.1-845: Bed level data only applied slightly

EddyMI3d, 2018-09-06:
Hi!
It may be, that this exact issue has been posted about before, but did not find one with this firmware version.

Here we go: I did all the Calibration and settings as recommended by the manual.
When it comes to the **First Layer Calibration**, it occurs, that the Z-axis tries to follow the unevenness of the bed (the spindles are rotating slightly), but not that much that it compensates the bed mesh level data.
Still, in some areas of the first layer the printed filament is plain flat, in some others it just adheres. 

It worked way much better with the MK2S with Firmware 3.1.0

slate0, 2018-11-09:
Might be related to this?
https://github.com/prusa3d/Prusa-Firmware/issues/1159

EddyMI3d, 2018-11-09:
Thanks for that hint!

closed 2023-09-05 by github-actions[bot]