prusa3d/Prusa-Firmware #1119: MK2 build profile issues

tkircher, 2018-09-08:
DEFAULT_MAX_FEEDRATE_SILENT and DEFAULT_MAX_ACCELERATION_SILENT are referenced in ConfigurationStore.cpp but not defined in the MK2 configuration files.

There's also a remnant lcd_print_at() in mesh_bed_calibration.cpp that could probably be replaced by a lcd_puts_at_P().

And SUPPORT_VERBOSITY needs to be undefined. But when looking into that, I noticed some potential logic errors:

                #ifdef SUPPORT_VERBOSITY
                if (verbosity_level >= 1)
                {
                    bool clamped = world2machine_clamp(current_position[X_AXIS], current_position[Y_AXIS]);
                        clamped ? SERIAL_PROTOCOLPGM("First calibration point clamped.\n") : SERIAL_PROTOCOLPGM("No clamping for first calibration point.\n");
                }
                #endif //SUPPORT_VERBOSITY

Here, world2machine_clamp() is only called if SUPPORT_VERBOSITY is defined. But further down,

                        world2machine_clamp(current_position[X_AXIS], current_position[Y_AXIS]);
                        #ifdef SUPPORT_VERBOSITY
                        if (verbosity_level >= 1) {

                                SERIAL_PROTOCOL(mesh_point);
                                clamped ? SERIAL_PROTOCOLPGM(": xy clamped.\n") : SERIAL_PROTOCOLPGM(": no xy clamping\n");
                        }
                        #endif // SUPPORT_VERBOSITY

This call is outside the #ifdef. Seems like one of these is not as intended. In the latter case, there's a compile error using the MK2 configuration because clamped isn't defined inside the scope.

tkircher, 2018-09-08:
Isn't there a plan to create a unified code base?

3d-gussner, 2018-09-09:
Yes it is but it looks like Prusa focus on mk3 and mk25 and mmu2 at this moment. Have so say that the mk2 firmware is stable and good.
Let's hope mk2s will get some attention soon.

BTW I really liked the G80 verbose mode but Prusa broke it somewhere and just disabled it in the mk25 and mk3 variants.

tkircher, 2018-09-09:
Thanks for your response.

There are plenty of us out here who have an interest in seeing the Prusa firmware cleaned up, and are willing to put some effort into it. I own a MK2 that I upgraded using a PINDA2 and a mechanical filament runout sensor. These various options, whether it's a MK42 or MK52 heat bed, whether there's any filament runout sensor, whether there are mechanical endstops, and so on, should all be configurable options, and every combination of options should at the very least compile successfully.

It makes sense for Prusa to prioritize the MK3 and MK2.5, and MMU2, of course. Those are their flagship products and they absolutely have to work. But the firmware is also a huge mess of spaghetti code. Cleaning it up is going to be essential to making it reliable in the long run, and part of that is going to require making sure every possible permutation of feature is supported.

If there were actually a clear roadmap for this project, with milestones and a detailed todo list, Prusa would not necessarily have to do it by themselves.

3d-gussner, 2018-09-09:
@tkircher Totally  agree. The MK2 branch was cleaned-up by Thess and now mkbel is pushing some requestes to cleanup the MK3 branch. It takes a while.

I got for my MK2 also a P.I.N.D.A.v2, bought a 12V Noctua fan and will add a Prusa laser filament runout sensor, as i think the MK42 bed is great and not worth to been put in the electronics-don't-know-when-i-will-use-it-again box.

So plan is to build a MK2.3 or in the future with bondtech and a 12V 5050 part rpm fan:
- MK2.1 = with Noctua fan
- MK2.2 = with Noctua fan and PINDAv2
- MK2.3 = with Noctua fan, PINDAv2 and Prusa 'laser' filament sensor
- MK2.4 = with Noctua fan, PINDAv2, Fil. sensor and bondtech gears
- MK2.45 = with Noctua fan, PINDAv2, Fil. sonsor, bondtech gears and 5050 12V rpm fan

It is a pity that Prusa broke the verbose mode of the G80 mesh bed leveling command as it was sometimes helpful to find issues.

closed 2023-09-05 by github-actions[bot]